ros2/kilted_tutorial_party #508: Testing

tfoote, 2025-05-01:
Check the documentation for the 'Testing' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Testing page](https://docs.ros.org/en/kilted/Installation/Testing.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

avanmalleghem, 2025-05-03:
Interested in !

avanmalleghem, 2025-05-03:
I run into the following issue but I guess it is because we are in a pre-release ? Otherwise, everything works as expected.
[image]

mjcarroll, 2025-05-09:
CC @nuclearsandwich is this related to key changes?

closed 2025-05-09 by mjcarroll