ros2/kilted_tutorial_party #511: The Keystore

tfoote, 2025-05-01:
Check the documentation for the 'The Keystore' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [The Keystore page](https://docs.ros.org/en/kilted/Tutorials/Advanced/Security/The-Keystore.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

Tenveis, 2025-05-04:
Hi, I'd like to work on this issue. Could you please assign it to me?

JatinVira, 2025-05-05:
Im working on this.
Thanks

fujitatomoya, 2025-05-07:
I think it should be installed via `rosdep` at the installation page, so that would be no problem. after installing numpy, does everything work without any problems? if you attach the screenshots, that would be really appreciated as confirmation.

JatinVira, 2025-05-07:
@fujitatomoya 
This is too advanced for me i would not be able to do this.
Remove me as the assigned
Thanks

fujitatomoya, 2025-05-07:
I do not see any problems during this procedure.

```console
root@tomoyafujita:~/sros2_demo# cd ~/sros2_demo/demo_keystore/public
root@tomoyafujita:~/sros2_demo/demo_keystore/public# openssl x509 -in ca.cert.pem -text -noout
Certificate:
    Data:
        Version: 3 (0x2)
        Serial Number:
            67:56:6e:89:84:03:e0:f0:62:56:0f:e7:7f:73:55:54:f0:32:c2:58
        Signature Algorithm: ecdsa-with-SHA256
        Issuer: CN = sros2CA
        Validity
            Not Before: May  6 17:08:26 2025 GMT
            Not After : May  5 17:08:26 2035 GMT
        Subject: CN = sros2CA
        Subject Public Key Info:
            Public Key Algorithm: id-ecPublicKey
                Public-Key: (256 bit)
                pub:
                    04:33:c3:0e:9a:3b:04:67:18:4c:f8:fc:ab:93:1f:
                    8c:d0:31:8b:45:70:29:9c:44:b3:33:c2:ae:ca:01:
                    59:2e:ce:7a:d6:26:71:ae:8b:98:ee:4d:b4:32:27:
                    92:89:33:4f:79:f0:a2:bf:26:65:80:69:8e:93:1f:
                    a4:8c:f5:a5:a8
                ASN1 OID: prime256v1
                NIST CURVE: P-256
        X509v3 extensions:
            X509v3 Basic Constraints: critical
                CA:TRUE, pathlen:1
    Signature Algorithm: ecdsa-with-SHA256
    Signature Value:
        30:45:02:20:0e:c0:d7:c1:34:99:f9:3c:14:8f:33:07:12:0c:
        1d:a0:ea:8e:bc:ad:de:b0:2a:41:fb:e4:03:3a:c7:88:3f:40:
        02:21:00:ee:e3:8f:e4:35:02:30:67:d0:a7:55:7d:e2:95:c7:
        c8:cb:28:ed:55:1a:09:80:11:94:22:75:16:f8:7c:53:bf

root@tomoyafujita:~/sros2_demo# cd ~/sros2_demo/demo_keystore/private
root@tomoyafujita:~/sros2_demo/demo_keystore/private# openssl ec -in ca.key.pem -text -noout
read EC key
Private-Key: (256 bit)
priv:
    28:f3:35:46:d9:ba:57:75:21:7b:4c:a4:95:43:df:
    df:24:05:f5:58:53:f6:75:1a:68:d1:5f:94:f5:80:
    93:e8
pub:
    04:33:c3:0e:9a:3b:04:67:18:4c:f8:fc:ab:93:1f:
    8c:d0:31:8b:45:70:29:9c:44:b3:33:c2:ae:ca:01:
    59:2e:ce:7a:d6:26:71:ae:8b:98:ee:4d:b4:32:27:
    92:89:33:4f:79:f0:a2:bf:26:65:80:69:8e:93:1f:
    a4:8c:f5:a5:a8
ASN1 OID: prime256v1
NIST CURVE: P-256
root@tomoyafujita:~/sros2_demo/demo_keystore/private# openssl smime -verify -in governance.p7s -CAfile ../public/permissions_ca.cert.pem
Can't open "governance.p7s" for reading, No such file or directory
8020E5CD437F0000:error:80000002:system library:BIO_new_file:No such file or directory:../crypto/bio/bss_file.c:67:calling fopen(governance.p7s, r)
8020E5CD437F0000:error:10000080:BIO routines:BIO_new_file:no such file:../crypto/bio/bss_file.c:75:

```

closed 2025-05-07 by fujitatomoya