ros2/kilted_tutorial_party #512: Time Travel With Tf2 Cpp

tfoote, 2025-05-01:
Check the documentation for the 'Time Travel With Tf2 Cpp' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Time Travel With Tf2 Cpp page](https://docs.ros.org/en/kilted/Tutorials/Intermediate/Tf2/Time-Travel-With-Tf2-Cpp.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

Aniketkumarroy, 2025-05-03:
Hi, I will take this up, Please assign me

Aniketkumarroy, 2025-05-03:
### **Following the static carrot position 5 seconds earlier**

[image]

- [x] I was able to follow the documentation.
- [x] The documentation seemed clear to me.
- [x] The documentation didn't have any obvious errors.

ahcorde, 2025-05-05:
Thank you for testing

closed 2025-05-05 by ahcorde