ros2/kilted_tutorial_party #513: Time Travel With Tf2 Py

tfoote, 2025-05-01:
Check the documentation for the 'Time Travel With Tf2 Py' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Time Travel With Tf2 Py page](https://docs.ros.org/en/kilted/Tutorials/Intermediate/Tf2/Time-Travel-With-Tf2-Py.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

Tenveis, 2025-05-04:
I wish to take up this, but it seems the External Link is not working.

[image]

JatinVira, 2025-05-05:
So basically for this issue,
the page for writing a `Tf2 Time Travel` using `Python` is not published
here's a snippet from the `intermediate` tutorials list on `Kilted`

[image]

Should we open up a PR to address this issue?

The equivalent code in c++ for this functionality is available

JatinVira, 2025-05-06:
@Yadunund 

Same problem for this issue.
It appears this is also an auto generated test, that should not exist

Yadunund, 2025-05-06:
Closing as this ticket was created due to an error in the requirements config. This has been fixed for future iterations: https://github.com/osrf/ros2_test_cases/pull/1560/commits/4e797d23bedc5cd8d4c4177a7bd16b621e09bf63

closed 2025-05-06 by Yadunund

JatinVira, 2025-05-07:
@ahcorde 
Please give the issue an assigne if possible. 
Thanks