ros2/kilted_tutorial_party #514: Topic executables in `demo_nodes_py`

tfoote, 2025-05-01:
## Setup
- Rmw: Cyclone
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check `listener` and `talker`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_py talker
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 run demo_nodes_py listener
       ```
  
    
  </details>
- [ ] **Check `listener` and `talker` QOS**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_py talker_qos
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 run demo_nodes_py listener_qos
       ```
  
    
  </details>
- [ ] **Check serialized `listener` and `talker`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_py talker
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 run demo_nodes_py listener_serialized
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

Innomer, 2025-05-04:
Can this issue be assigned to me please? Thank you.

seekbot, 2025-05-06:
I can do this.

seekbot, 2025-05-06:
Here is the test recording:
[image]