ros2/kilted_tutorial_party #515: Topic executables in `demo_nodes_py`

tfoote, 2025-05-01:
## Setup
- Rmw: Cyclone
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Checks
- [ ] **Check `listener` and `talker`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_py talker
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 run demo_nodes_py listener
       ```
  
    
  </details>
- [ ] **Check `listener` and `talker` QOS**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_py talker_qos
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 run demo_nodes_py listener_qos
       ```
  
    
  </details>
- [ ] **Check serialized `listener` and `talker`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_py talker
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 run demo_nodes_py listener_serialized
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

suchetanrs, 2025-05-02:
- [x] Check listener and talker

[image]

- [x] Check listener and talker QOS

[image]

- [x] Check serialized listener and talker

[image]

suchetanrs, 2025-05-02:
[image]

System information

closed 2025-05-02 by ahcorde