ros2/kilted_tutorial_party #516: Topic executables in `demo_nodes_py`

tfoote, 2025-05-01:
## Setup
- Rmw: Fastdds
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check `listener` and `talker`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_py talker
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 run demo_nodes_py listener
       ```
  
    
  </details>
- [ ] **Check `listener` and `talker` QOS**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_py talker_qos
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 run demo_nodes_py listener_qos
       ```
  
    
  </details>
- [ ] **Check serialized `listener` and `talker`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_py talker
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 run demo_nodes_py listener_serialized
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

JasperGrant, 2025-05-02:
I am interested to test this one. Can I please have it assigned?

JasperGrant, 2025-05-02:
Test 1:
  Talker: 
  `[INFO] [1746196475.216715900] [talker]: Publishing: "Hello World: 0"`
  Listener: 
  `[INFO] [1746196475.502011700] [listener]: I heard: [Hello World: 0]`

Test 2:
  Talker: 
  `[INFO] [1746196729.744329800] [talker_qos]: Best effort talker
  [INFO] [1746196730.781914200] [talker_qos]: Publishing: "Hello World: 0"`
  Listener:
  `[INFO] [1746196725.998940100] [listener_qos]: Best effort listener
  [INFO] [1746196730.790494200] [listener_qos]: I heard: [Hello World: 0]]`

Test 3:
  Talker:
  `[INFO] [1746196911.684873000] [talker]: Publishing: "Hello World: 0"`
  Listener:
  `[INFO] [1746196911.935031200] [serialized_subscriber]: I heard: "b'\x00\x01\x00\x00\x0f\x00\x00\x00Hello World: 0\x00\x00'"`

All tests passed. Ready to be closed!

ahcorde, 2025-05-02:
Thank you for testing

closed 2025-05-02 by ahcorde