ros2/kilted_tutorial_party #519: Topic monitor

tfoote, 2025-05-01:
## Setup
- Rmw: Cyclone
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Links
- [Reference URL](https://github.com/ros2/demos/blob/kilted/topic_monitor/README.md)

## Checks
- [ ] **Comparing reliability QoS settings**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       You will need two machines with ROS 2: one mobile and one stationary.
       
       
    1.
       ```bash
       # User input in terminal 1
       ros2 run topic_monitor topic_monitor --display
       ```
    1.
       Run the `ros2 launch topic_monitor reliability_demo_launch.py` executable on the stationary machine. This will start two nodes: one publishing in “reliable” mode, and one in “best effort”.
       
       ```bash
       # User input in terminal 1
       ros2 launch topic_monitor reliability_demo_launch.py
       ```
    1.
       Start the monitor on a mobile machine such as a laptop. Use `ros2 run topic_monitor topic_monitor --display --allowed-latency 5` to account for any latency that may occur re-sending the reliable messages.
       
       ```bash
       # User input in terminal 1
       ros2 run topic_monitor topic_monitor --display --allowed-latency 5
       ```
    1.
       Take the mobile machine out of range of the monitor, and observe how the reception rates differ for the different topics.
       
  
    **Expect**
    
    1.
       You should see that the "reliable" topic has a reception rate that is almost always either 0 or 100%, while the "best effort" topic has a reception rate that fluctuates based on the strength of the connection.
       
       [image]
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

Innomer, 2025-05-04:
Can I be assigned this Issue?

seekbot, 2025-05-06:
Interested @ahcorde

mjcarroll, 2025-05-09:
@seekbot any update?

seekbot, 2025-05-09:
Hi, 

I didn't really get the chance to go out of network range (I will follow up soon). One thing to note was that the binary does not have 'matplotlib' package installed, so I had to run 'pip install matplotlib' to get the graph.

For now, here are video demos showcasing step i to iv:
- Stationary (Windows 11 Laptop): https://youtu.be/uuzLLtXKbcA
- Mobile (Windows 10 Laptop): https://youtu.be/DZRDSDrl8Vc

Tanishq30052002, 2025-05-12:
@mjcarroll 
Stationary Machine

[image]

Mobile Machine

[image]

ERROR: While plotting the plots, the matplotlib window doesn't respond (Even nothing gets printed)