ros2/kilted_tutorial_party #52: Check ROS 2 launch CLI

tfoote, 2025-05-01:
## Setup
- Rmw: Zenoh
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Checks
- [ ] **Check launch package**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch demo_nodes_cpp talker_listener_launch.py
       ```
  
    **Expect**
    
    1.
       You should see something like the following
       ```bash
       # stdout in terminal 1
       [INFO] [launch]: All log files can be found below /home/parallels/.ros/log/2022-03-21-17-38-54-737866-parallels-Parallels-Virtual-Platform-3717
       [INFO] [launch]: Default logging verbosity is set to INFO
       [INFO] [talker-1]: process started with pid [3722]
       [INFO] [listener-2]: process started with pid [3724]
       [talker-1] [INFO] [1647909535.937731548] [talker]: Publishing: 'Hello World: 1'
       [listener-2] [INFO] [1647909535.938337387] [listener]: I heard: [Hello World: 1]
       [talker-1] [INFO] [1647909536.937932068] [talker]: Publishing: 'Hello World: 2'
       [listener-2] [INFO] [1647909536.938125486] [listener]: I heard: [Hello World: 2]
       [talker-1] [INFO] [1647909537.936162331] [talker]: Publishing: 'Hello World: 3'
       [listener-2] [INFO] [1647909537.936366391] [listener]: I heard: [Hello World: 3]
       ...
       ```
  </details>
- [ ] **Check launch launch file directly**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       Download a launch file, for example with `curl`.
       ```bash
       # User input in terminal 1
       curl https://raw.githubusercontent.com/ros2/demos/master/demo_nodes_cpp/launch/topics/talker_listener_launch.py > /tmp/talker_listener_launch.py
       ```
    1.
       ```bash
       # User input in terminal 1
       ros2 launch /tmp/talker_listener_launch.py
       ```
  
    **Expect**
    
    1.
       You should see something like the following
       ```bash
       # stdout in terminal 1
       [INFO] [launch]: All log files can be found below /home/parallels/.ros/log/2022-03-21-17-38-54-737866-parallels-Parallels-Virtual-Platform-3717
       [INFO] [launch]: Default logging verbosity is set to INFO
       [INFO] [talker-1]: process started with pid [3722]
       [INFO] [listener-2]: process started with pid [3724]
       [talker-1] [INFO] [1647909535.937731548] [talker]: Publishing: 'Hello World: 1'
       [listener-2] [INFO] [1647909535.938337387] [listener]: I heard: [Hello World: 1]
       [talker-1] [INFO] [1647909536.937932068] [talker]: Publishing: 'Hello World: 2'
       [listener-2] [INFO] [1647909536.938125486] [listener]: I heard: [Hello World: 2]
       [talker-1] [INFO] [1647909537.936162331] [talker]: Publishing: 'Hello World: 3'
       [listener-2] [INFO] [1647909537.936366391] [listener]: I heard: [Hello World: 3]
       ...
       ```
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

jncfa, 2025-05-02:
Tested with Ubuntu Docker Container for amd64(under Rosetta 2) (ghcr.io/sloretz/ros-testing:kilted-desktop)

[image]

ahcorde, 2025-05-02:
thank you for testing