ros2/kilted_tutorial_party #520: Topic monitor

tfoote, 2025-05-01:
## Setup
- Rmw: Cyclone
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Reference URL](https://github.com/ros2/demos/blob/kilted/topic_monitor/README.md)

## Checks
- [ ] **Comparing reliability QoS settings**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       You will need two machines with ROS 2: one mobile and one stationary.
       
       
    1.
       ```bash
       # User input in terminal 1
       ros2 run topic_monitor topic_monitor --display
       ```
    1.
       Run the `ros2 launch topic_monitor reliability_demo_launch.py` executable on the stationary machine. This will start two nodes: one publishing in “reliable” mode, and one in “best effort”.
       
       ```bash
       # User input in terminal 1
       ros2 launch topic_monitor reliability_demo_launch.py
       ```
    1.
       Start the monitor on a mobile machine such as a laptop. Use `ros2 run topic_monitor topic_monitor --display --allowed-latency 5` to account for any latency that may occur re-sending the reliable messages.
       
       ```bash
       # User input in terminal 1
       ros2 run topic_monitor topic_monitor --display --allowed-latency 5
       ```
    1.
       Take the mobile machine out of range of the monitor, and observe how the reception rates differ for the different topics.
       
  
    **Expect**
    
    1.
       You should see that the "reliable" topic has a reception rate that is almost always either 0 or 100%, while the "best effort" topic has a reception rate that fluctuates based on the strength of the connection.
       
       [image]
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

zamantousif, 2025-05-04:
Happy to test this. Please assign this issue to me.

zamantousif, 2025-05-04:
**Tested with below settings:**
Machine 1 (mobile) running ROS2 Kilted with CycloneDDS inside docker image:
`docker run --rm=true -ti --net=host --env ROS_DOMAIN_ID=4 ghcr.io/sloretz/ros-testing:kilted-desktop bash`

Machine 2 (stationary) running ROS2 Humble with CycloneDDS on host.
**Test results:**

[image]

ahcorde, 2025-05-05:
Thank you for testing

closed 2025-05-05 by ahcorde