ros2/kilted_tutorial_party #521: Topic monitor

tfoote, 2025-05-01:
## Setup
- Rmw: Fastdds
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Links
- [Reference URL](https://github.com/ros2/demos/blob/kilted/topic_monitor/README.md)

## Checks
- [ ] **Comparing reliability QoS settings**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       You will need two machines with ROS 2: one mobile and one stationary.
       
       
    1.
       ```bash
       # User input in terminal 1
       ros2 run topic_monitor topic_monitor --display
       ```
    1.
       Run the `ros2 launch topic_monitor reliability_demo_launch.py` executable on the stationary machine. This will start two nodes: one publishing in “reliable” mode, and one in “best effort”.
       
       ```bash
       # User input in terminal 1
       ros2 launch topic_monitor reliability_demo_launch.py
       ```
    1.
       Start the monitor on a mobile machine such as a laptop. Use `ros2 run topic_monitor topic_monitor --display --allowed-latency 5` to account for any latency that may occur re-sending the reliable messages.
       
       ```bash
       # User input in terminal 1
       ros2 run topic_monitor topic_monitor --display --allowed-latency 5
       ```
    1.
       Take the mobile machine out of range of the monitor, and observe how the reception rates differ for the different topics.
       
  
    **Expect**
    
    1.
       You should see that the "reliable" topic has a reception rate that is almost always either 0 or 100%, while the "best effort" topic has a reception rate that fluctuates based on the strength of the connection.
       
       [image]
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

Poo-09, 2025-05-02:
I'm interested to test, please assign this ticket to me

ShahazadAbdulla, 2025-05-06:
You can always do the test, You don't particularly need to wait to be assigned, if you are dropping this then I can take it or if you wanna do feel free to. 
Thank You

mjcarroll, 2025-05-09:
@Poo-09 any update? Have you completed this ticket?

ShahazadAbdulla, 2025-05-09:
@mjcarroll If @Poo-09  is not responding I would be happy to take on the issue  and would be good if you assigned me to this. If @Poo-09 is working then no problem. Happy to help.

Aniketkumarroy, 2025-05-12:
Hi I tested this
Machine 1

[image]

Machine 2

[image]