ros2/kilted_tutorial_party #523: Topic monitor

tfoote, 2025-05-01:
## Setup
- Rmw: Zenoh
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Reference URL](https://github.com/ros2/demos/blob/kilted/topic_monitor/README.md)

## Checks
- [ ] **Comparing reliability QoS settings**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       You will need two machines with ROS 2: one mobile and one stationary.
       
       
    1.
       ```bash
       # User input in terminal 1
       ros2 run topic_monitor topic_monitor --display
       ```
    1.
       Run the `ros2 launch topic_monitor reliability_demo_launch.py` executable on the stationary machine. This will start two nodes: one publishing in “reliable” mode, and one in “best effort”.
       
       ```bash
       # User input in terminal 1
       ros2 launch topic_monitor reliability_demo_launch.py
       ```
    1.
       Start the monitor on a mobile machine such as a laptop. Use `ros2 run topic_monitor topic_monitor --display --allowed-latency 5` to account for any latency that may occur re-sending the reliable messages.
       
       ```bash
       # User input in terminal 1
       ros2 run topic_monitor topic_monitor --display --allowed-latency 5
       ```
    1.
       Take the mobile machine out of range of the monitor, and observe how the reception rates differ for the different topics.
       
  
    **Expect**
    
    1.
       You should see that the "reliable" topic has a reception rate that is almost always either 0 or 100%, while the "best effort" topic has a reception rate that fluctuates based on the strength of the connection.
       
       [image]
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

zamantousif, 2025-05-04:
Since I already worked on #520, I could also test this ticket. Please feel free to assign this to me.

zamantousif, 2025-05-09:
Okay, sorry for the late response. I was stuck in getting the two machines to talk to each other with rmw_zenoh. However, it seems to work now.
I got the following results when running the topic monitor in different hosts with the zenoh router of the listener connect to the endpoint of the other host running the reliability demo.

[image]

Though the instructions say, `"reliable" topic has a reception rate that is almost always either 0 or 100%`, I got the following response when away from the network and for a brief moment it seems that the reliable topic was in between 0 and 100% otherwise, the test runs fine.

[image]

closed 2025-05-09 by ahcorde