ros2/kilted_tutorial_party #524: Topic Statistics Tutorial

tfoote, 2025-05-01:
Check the documentation for the 'Topic Statistics Tutorial' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Topic Statistics Tutorial page](https://docs.ros.org/en/kilted/Tutorials/Advanced/Topic-Statistics-Tutorial/Topic-Statistics-Tutorial.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

avanmalleghem, 2025-05-02:
Interested in

avanmalleghem, 2025-05-03:
PR opened here : https://github.com/ros2/ros2_documentation/pull/5495

ahcorde, 2025-05-05:
PR is ros2 documentation is merged. Thank you for testing

closed 2025-05-05 by ahcorde