ros2/kilted_tutorial_party #526: Ubuntu Install Binary

tfoote, 2025-05-01:
Check the documentation for the 'Ubuntu Install Binary' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Ubuntu Install Binary page](https://docs.ros.org/en/kilted/Installation/Alternatives/Ubuntu-Install-Binary.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

Juliaj, 2025-05-01:
Wasn't able to install the Binary by following the instructions from https://docs.ros.org/en/kilted/Installation/Alternatives/Ubuntu-Install-Binary.html 

Steps failed: [Enable required repositories](https://docs.ros.org/en/kilted/Installation/Alternatives/Ubuntu-Install-Binary.html#id4)[](https://docs.ros.org/en/kilted/Installation/Alternatives/Ubuntu-Install-Binary.html#enable-required-repositories)

Error:
```
➜  ~ sudo apt update && sudo apt install curl -y
curl -o /tmp/ros2-testing-apt-source.deb https://ftp.osuosl.org/pub/ros/packages.ros.org/ros2-testing/ubuntu/pool/main/r/ros-apt-source/ros2-testing-apt-source_1.0.0\~$\(. /etc/os-release && echo $VERSION_CODENAME)_all.deb
sudo apt install /tmp/ros2-testing-apt-source.deb
zsh: parse error near `)'
```

<details>

<summary>Full console output</summary>

```
LANG=en_US.UTF-8
LANGUAGE=
LC_CTYPE="en_US.UTF-8"
LC_NUMERIC="en_US.UTF-8"
LC_TIME="en_US.UTF-8"
LC_COLLATE="en_US.UTF-8"
LC_MONETARY="en_US.UTF-8"
LC_MESSAGES="en_US.UTF-8"
LC_PAPER="en_US.UTF-8"
LC_NAME="en_US.UTF-8"
LC_ADDRESS="en_US.UTF-8"
LC_TELEPHONE="en_US.UTF-8"
LC_MEASUREMENT="en_US.UTF-8"
LC_IDENTIFICATION="en_US.UTF-8"
LC_ALL=en_US.UTF-8

➜  ~ sudo apt install software-properties-common
sudo add-apt-repository universe
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
software-properties-common is already the newest version (0.99.49.2).
0 upgraded, 0 newly installed, 0 to remove and 360 not upgraded.
Adding component(s) 'universe' to all repositories.
Press [ENTER] to continue or Ctrl-c to cancel.
Hit:1 http://packages.osrfoundation.org/gazebo/ubuntu-stable noble InRelease
Hit:2 https://download.docker.com/linux/ubuntu noble InRelease   
Hit:3 https://apt.foxglove.dev/studio stable InRelease           
Hit:4 http://apt.pop-os.org/proprietary noble InRelease          
Hit:5 http://apt.pop-os.org/release noble InRelease              
Hit:6 http://packages.ros.org/ros2/ubuntu noble InRelease        
Hit:7 http://apt.pop-os.org/ubuntu noble InRelease               
Hit:8 https://dl.google.com/linux/chrome/deb stable InRelease
Hit:9 http://apt.pop-os.org/ubuntu noble-security InRelease
Hit:10 http://apt.pop-os.org/ubuntu noble-updates InRelease
Hit:11 http://apt.pop-os.org/ubuntu noble-backports InRelease
Reading package lists... Done

➜  ~ sudo apt update && sudo apt install curl -y
curl -o /tmp/ros2-testing-apt-source.deb https://ftp.osuosl.org/pub/ros/packages.ros.org/ros2-testing/ubuntu/pool/main/r/ros-apt-source/ros2-testing-apt-source_1.0.0\~$\(. /etc/os-release && echo $VERSION_CODENAME)_all.deb
sudo apt install /tmp/ros2-testing-apt-source.deb
zsh: parse error near `)'

➜  ~ cat /etc/os-release
NAME="Pop!_OS"
VERSION="24.04 LTS"
ID=pop
ID_LIKE="ubuntu debian"
PRETTY_NAME="Pop!_OS 24.04 LTS"
VERSION_ID="24.04"
HOME_URL="https://pop.system76.com"
SUPPORT_URL="https://support.system76.com"
BUG_REPORT_URL="https://github.com/pop-os/pop/issues"
PRIVACY_POLICY_URL="https://system76.com/privacy"
VERSION_CODENAME=noble
UBUNTU_CODENAME=noble
LOGO=distributor-logo-pop-os
```
</details>

christophebedard, 2025-05-03:
Thanks for testing!

> Observation: seeing following when running step of "[Install dependencies using rosdep](https://docs.ros.org/en/kilted/Installation/Alternatives/Ubuntu-Install-Binary.html#id8)"
> 
> ```
> RROR: the following packages/stacks could not have their rosdep keys resolved
> to system dependencies:
> rosidl_dynamic_typesupport_fastrtps: Cannot locate rosdep definition for [fastdds]
> rmw_fastrtps_dynamic_cpp: Cannot locate rosdep definition for [fastdds]
> rmw_fastrtps_cpp: Cannot locate rosdep definition for [fastdds]
> rmw_fastrtps_shared_cpp: Cannot locate rosdep definition for [fastdds]
> ```

Can you try again and change `fastrtps` to `fastdds` in the `rosdep install` command? See https://github.com/ros2/ros2_documentation/pull/5482.

> Wasn't able to install the Binary by following the instructions from https://docs.ros.org/en/kilted/Installation/Alternatives/Ubuntu-Install-Binary.html from `zsh` shell. Workaround is to switch to `bash` shell or manually remove the escape char `\` added when copying/pasting the commands to zsh shell. See [ros2/ros2_documentation#5462](https://github.com/ros2/ros2_documentation/issues/5462)
> 
> Steps failed: [Enable required repositories](https://docs.ros.org/en/kilted/Installation/Alternatives/Ubuntu-Install-Binary.html#id4)[](https://docs.ros.org/en/kilted/Installation/Alternatives/Ubuntu-Install-Binary.html#enable-required-repositories)

Discussed in https://github.com/ros2/ros2_documentation/issues/5462.

christophebedard, 2025-05-03:
Thanks for confirming!

closed 2025-05-03 by christophebedard