ros2/kilted_tutorial_party #528: Understanding real-time programming

tfoote, 2025-05-01:
Check the documentation for the 'Understanding real-time programming' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Understanding real-time programming page](https://docs.ros.org/en/kilted/Tutorials/Demos/Real-Time-Programming.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

JatinVira, 2025-05-02:
Happy to take this Issue.
Please assign to me

JatinVira, 2025-05-03:
### Setup 
---
```
   PLATFORM INFORMATION
system           : Linux
processor        : x86_64

   RMW MIDDLEWARE
middleware name    : rmw_fastrtps_cpp

   ROS 2 INFORMATION
distribution name      : kilted
distribution type      : ros2
distribution status    : pre-release
release platforms      : {'debian': ['bookworm'], 'rhel': ['9'], 'ubuntu': ['noble']}
```

JatinVira, 2025-05-03:
@ahcorde 

[image]

the `Install and run the demo` section clearly mentions that this is supported only for `Connext` DDS.
However, the issue talks about Testing out `FastDDS`

should this part be skipped?

JatinVira, 2025-05-04:
@ahcorde ?

ahcorde, 2025-05-05:
@tfoote @mjcarroll ?

JatinVira, 2025-05-06:
@Yadunund ?

fujitatomoya, 2025-05-07:
@JatinVira i think label and issue description is wrong, because i cannot find the similar issue for any other RMW implementaions.

fujitatomoya, 2025-05-07:
@JatinVira can you please proceed with ConnextDDS instead? i just updated label and issue dsecription.

JatinVira, 2025-05-07:
Sure let me try this out

JatinVira, 2025-05-07:
[image]

[image]

JatinVira, 2025-05-08:
@fujitatomoya ?

JatinVira, 2025-05-09:
I want to have this issue highlighted out please.

fujitatomoya, 2025-05-09:
@JatinVira i will have a look when i can allocate some time.

fujitatomoya, 2025-05-09:
i think you already have `source`d the kilted workspace and build rmw_connextdds? i think what it tells us to do is that, we can just do the source build and install ConnextDDS dependent packages, then try the demonstration. the doc is NOT well-described for this procedure, i think i can take this over.

fujitatomoya, 2025-05-09:
the procedure works, but not well described for user tutorial. https://github.com/ros2/ros2_documentation/pull/5600 addresses this documentation quality issue.

```console
root@tomoyafujita:~/ros2_kilted# ros2 run pendulum_control pendulum_demo
Initial major pagefaults: 0
Initial minor pagefaults: 2139513
No results filename given, not writing results
rttest statistics:
  - Minor pagefaults: 0
  - Major pagefaults: 0
  Latency (time after deadline was missed):
    - Min: 2085 ns
    - Max: 29250 ns
    - Mean: 11871.763000 ns
    - Standard deviation: 3542.708907

```

since all procedure works without any problems (except documentation clarification), i will go ahead to close this issue.

@JatinVira thanks for testing this.

closed 2025-05-09 by fujitatomoya