ros2/kilted_tutorial_party #53: Check ROS 2 Multicast CLI

tfoote, 2025-05-01:
## Setup
- Rmw: Cyclone
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check send and receive**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 multicast receive
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 multicast send
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 1
       Waiting for UDP multicast datagram...
       Received from 10.211.55.9:59527: 'Hello World!'
       ```
    1.
       ```bash
       # stdout in terminal 2
       Sending one UDP multicast datagram...
       ```
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

ShahazadAbdulla, 2025-05-04:
I'll test this on Windows 11 / amd64 / binary / CycloneDDS

ShahazadAbdulla, 2025-05-04:
Successfully completed checks for Issue #52 on Windows 11 / amd64 / binary installation using CycloneDDS (`rmw_cyclonedds_cpp`).

- Started `ros2 multicast receive` in Terminal 1.
- Ran `ros2 multicast send` in Terminal 2.
- Terminal 1 successfully printed the `Received from <IP>:<Port>: 'Hello World!'` message as expected.

All checks passed.

closed 2025-05-05 by ahcorde