ros2/kilted_tutorial_party #533: Understanding ROS2 Topics

tfoote, 2025-05-01:
Check the documentation for the 'Understanding ROS2 Topics' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Understanding ROS2 Topics page](https://docs.ros.org/en/kilted/Tutorials/Beginner-CLI-Tools/Understanding-ROS2-Topics/Understanding-ROS2-Topics.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

khaledgabr77, 2025-05-01:
* [x]  I was able to follow the documentation.
* [x]  The documentation seemed clear to me.
* [x]  The documentation didn't have any obvious errors.

[image]

[image]

[image]

akky20, 2025-05-04:
@khaledgabr77, we request you to first let the maintainer assign you the issue before testing, as you have tested many issues without being assigned or commenting that you're interested.

khaledgabr77, 2025-05-04:
To @tfoote @ahcorde ,

When the `kilted_tutorial_party` was released, I happened to be on holiday but still did my best to contribute where I could. In some cases, I mentioned that I was working on an issues(https://github.com/ros2/kilted_tutorial_party/issues/549) (https://github.com/ros2/kilted_tutorial_party/issues/518) (https://github.com/ros2/kilted_tutorial_party/issues/492) and so on, though I admit I may have missed commenting on others. My sole intention was to support and help improve the project.

I’ve also noticed that several issues were closed without prior assignment or contributor comments, so I wasn’t aware that being assigned first was a strict requirement. If that’s now the expected workflow, I’m happy to follow it going forward.

To @akky20 , i don't know what you mean by `we..`. I’d appreciate it if we could keep the communication constructive—we’re all here to collaborate and make this project better together.

akky20, 2025-05-04:
Hey @khaledgabr77 , I think you misderstood my message, Just to clarify, my earlier message wasn’t meant to be rude. The reason I pointed it out is that, as per the contribution guidelines for this tutorial party, it's important that contributors first get assigned to an issue before starting work. This helps avoid overlap and keeps things organized, especially with so many people participating.
When I said “we,” I was referring to all the contributor around the world who are contributing to this tutorial party. We’re all here to collaborate, and I’m glad you’re on board with following the expected process going forward. Looking forward to your continued contributions mate!

ahcorde, 2025-05-05:
Thank you for testing

closed 2025-05-05 by ahcorde