ros2/kilted_tutorial_party #536: Using Parameters In A Class Python

tfoote, 2025-05-01:
Check the documentation for the 'Using Parameters In A Class Python' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Using Parameters In A Class Python page](https://docs.ros.org/en/kilted/Tutorials/Beginner-Client-Libraries/Using-Parameters-In-A-Class-Python.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

Tenveis, 2025-05-02:
I am interested.

Tenveis, 2025-05-02:
- [X] I was able to follow the documentation.
- [X] The documentation seemed clear to me.
- [X] The documentation didn't have any obvious errors.

In Section 2.1.1, since we have imported the **rcl_interfaces** module. It might be helpful to mention that  we need to add it as a dependency in the **package.xml** file to ensure there is no dependency issue. 

[image]

Tenveis, 2025-05-03:
sure.

christophebedard, 2025-05-03:
Closing this as the docs PR was created and the instructions are otherwise correct: https://github.com/ros2/ros2_documentation/pull/5514

Thanks for testing!

closed 2025-05-03 by christophebedard