ros2/kilted_tutorial_party #539: Using Stamped Datatypes With Tf2 Ros MessageFilter

tfoote, 2025-05-01:
Check the documentation for the 'Using Stamped Datatypes With Tf2 Ros MessageFilter' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Using Stamped Datatypes With Tf2 Ros MessageFilter page](https://docs.ros.org/en/kilted/Tutorials/Intermediate/Tf2/Using-Stamped-Datatypes-With-Tf2-Ros-MessageFilter.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

Tanishq30052002, 2025-05-03:
Tested and Validated
Working

- I was able to follow the documentation.
- The documentation seemed clear to me.
- The documentation didn't have any obvious errors.

[image]
[image]

Tanishq30052002, 2025-05-03:
- I was able to follow the documentation.
- The documentation seemed clear to me.
- The documentation didn't have any obvious errors (Suggestions)
   - Instructions to add the launch file in setup.py.
     - These lines instruct how to add the launch file to the setup.py
      ``` py
      from glob import glob
      # add the below line to data files
      ('share/' + package_name + '/launch', glob('launch/*.py')),
     ```
  - Instruction to install [turtle_tf2_py](https://index.ros.org/p/turtle_tf2_py)
    - Below cmd should work.
      ``` bash
      sudo apt install -y ros-kilted-turtle-tf2-py
      ```

Other than the documentation implementation is working fine. 

[image]

ahcorde, 2025-05-05:
Do you mind to open a PR in ros2_documentation with your suggestions?

closed 2025-05-08 by ahcorde