ros2/kilted_tutorial_party #540: Using URDF with Robot State Publisher

tfoote, 2025-05-01:
Check the documentation for the 'Using URDF with Robot State Publisher' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Using URDF with Robot State Publisher page](https://docs.ros.org/en/kilted/Tutorials/Intermediate/URDF/Using-URDF-with-Robot-State-Publisher.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

akky20, 2025-05-02:
Interested

akky20, 2025-05-04:
Done with the testing...
No issues

https://github.com/user-attachments/assets/ff5769ce-909a-4c80-ae19-7e8290dd3241

akky20, 2025-05-05:
The launch file is runnign perfectly fine , but the problem is with rviz and I think everyone else also is facing the same problem the Icon one, as confirmed from other Rviz issue tickets.

[image]

akky20, 2025-05-07:
Solve the rviz issue and tested working fine !!

https://github.com/user-attachments/assets/1e393941-0489-4870-913b-b511be1dabcc

closed 2025-05-07 by ahcorde