ros2/kilted_tutorial_party #541: Using Xacro to Clean Up a URDF File

tfoote, 2025-05-01:
Check the documentation for the 'Using Xacro to Clean Up a URDF File' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Using Xacro to Clean Up a URDF File page](https://docs.ros.org/en/kilted/Tutorials/Intermediate/URDF/Using-Xacro-to-Clean-Up-a-URDF-File.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

jncfa, 2025-05-02:
I used the urdf_tutorial:
`xacro src/urdf_tutorial/urdf/08-macroed.urdf.xacro | diff src/urdf_tutorial/urdf/08-macroed.urdf.xacro - `

This is the output of the diff between the .xacro and the generated urdf (which seems to be basically what was expected):
<details>
<summary>Diff between xacro and URDF</summary>

```diff
1,10c1,6
< <?xml version="1.0"?>
< <robot name="macroed" xmlns:xacro="http://ros.org/wiki/xacro">
< 
<   <xacro:property name="width" value="0.2" />
<   <xacro:property name="leglen" value="0.6" />
<   <xacro:property name="polelen" value="0.2" />
<   <xacro:property name="bodylen" value="0.6" />
<   <xacro:property name="baselen" value="0.4" />
<   <xacro:property name="wheeldiam" value="0.07" />
< 
---
> <?xml version="1.0" ?>
> <!-- =================================================================================== -->
> <!-- |    This document was autogenerated by xacro from src/urdf_tutorial/urdf/08-macroed.urdf.xacro | -->
> <!-- |    EDITING THIS FILE BY HAND IS NOT RECOMMENDED                                 | -->
> <!-- =================================================================================== -->
> <robot name="macroed">
14d9
< 
18d12
< 
22,29d15
< 
<   <xacro:macro name="default_inertial" params="mass">
<     <inertial>
<       <mass value="${mass}" />
<       <inertia ixx="1e-3" ixy="0.0" ixz="0.0" iyy="1e-3" iyz="0.0" izz="1e-3" />
<     </inertial>
<   </xacro:macro>
< 
33c19
<         <cylinder radius="${width}" length="${bodylen}"/>
---
>         <cylinder length="0.6" radius="0.2"/>
39c25,115
<         <cylinder radius="${width}" length="${bodylen}"/>
---
>         <cylinder length="0.6" radius="0.2"/>
>       </geometry>
>     </collision>
>     <inertial>
>       <mass value="10"/>
>       <inertia ixx="1e-3" ixy="0.0" ixz="0.0" iyy="1e-3" iyz="0.0" izz="1e-3"/>
>     </inertial>
>   </link>
>   <link name="right_leg">
>     <visual>
>       <geometry>
>         <box size="0.6 0.1 0.2"/>
>       </geometry>
>       <origin rpy="0 1.5707963267948966 0" xyz="0 0 -0.3"/>
>       <material name="white"/>
>     </visual>
>     <collision>
>       <geometry>
>         <box size="0.6 0.1 0.2"/>
>       </geometry>
>       <origin rpy="0 1.5707963267948966 0" xyz="0 0 -0.3"/>
>     </collision>
>     <inertial>
>       <mass value="10"/>
[... 1086 lines cut ...]
    <visual>
      <geometry>
        <box size="0.08 0.08 0.08"/>
      </geometry>
      <material name="blue"/>
      <origin xyz="-0.04 0 0"/>
    </visual>
    <collision>
      <geometry>
        <box size="0.08 0.08 0.08"/>
      </geometry>
    </collision>
    <inertial>
      <mass value="1"/>
      <inertia ixx="1e-3" ixy="0.0" ixz="0.0" iyy="1e-3" iyz="0.0" izz="1e-3"/>
    </inertial>
  </link>
  <joint name="tobox" type="fixed">
    <parent link="head"/>
    <child link="box"/>
    <origin xyz="0.1814 0 0.1414"/>
  </joint>
</robot>
```
</details>

mjcarroll, 2025-05-09:
> The documentation was for the most part clear, but I think it should include instructions to install xacro, just a simple:
apt-get install -y ros-kilted-xacro should suffice.

Can you open a PR for this on ros2/ros2_documentation?

closed 2025-05-09 by mjcarroll