ros2/kilted_tutorial_party #542: Wait for acknowledgment

tfoote, 2025-05-01:
Check the documentation for the 'Wait for acknowledgment' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Wait for acknowledgment page](https://docs.ros.org/en/kilted/Tutorials/Demos/Wait-for-Acknowledgment.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

jncfa, 2025-05-01:
Can I get assigned this one?

jncfa, 2025-05-01:
Setup: Docker Container w/ Rosetta 2 (ghcr.io/sloretz/ros-testing:kilted-desktop)

The documentation seems clear to me, but something about the demo with fastdds (both in arm64 and amd64) is not working.

[image]

mjcarroll, 2025-05-09:
https://github.com/ros2/rmw_fastrtps/issues/822