ros2/kilted_tutorial_party #543: Windows Development Setup

tfoote, 2025-05-01:
Check the documentation for the 'Windows Development Setup' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Windows Development Setup page](https://docs.ros.org/en/kilted/Installation/Alternatives/Windows-Development-Setup.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

purohitprachi72, 2025-05-03:
Hi, can I do this one?

Yadunund, 2025-05-05:
@JatinVira thanks for pointing that point. I've updated the ticket.

Tanishq30052002, 2025-05-08:
I am testing this out

Tanishq30052002, 2025-05-08:
We should add instruction to use administrator mode for setting long paths
[image]

Tanishq30052002, 2025-05-08:
@Yadunund windows 10 is about to come to an eol, so shouldnt we test for windows 11?

Tanishq30052002, 2025-05-08:
@ahcorde https://github.com/ros2/ros2/blob/kilted/pixi.toml

name should needs to be updated in the pixi.toml for kilted.
Should i create the pr for the same?

Tanishq30052002, 2025-05-08:
Working on Win 11
[image]

Tanishq30052002, 2025-05-08:
@ahcorde can we close it now?

ahcorde, 2025-05-09:
Thank you for testing

closed 2025-05-09 by ahcorde

Tanishq30052002, 2025-05-09:
@ahcorde although the issue closed, I have a suggestion to add in the docs, related to supported version of windows, since 10 is officially supported, but I have tried it on **win 11**, and there too it worked, so shouldn't we update the same? Also, Increasing path length size require administrator Control.