ros2/kilted_tutorial_party #544: Windows Install Binary

tfoote, 2025-05-01:
Check the documentation for the 'Windows Install Binary' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Windows Install Binary page](https://docs.ros.org/en/kilted/Installation/Windows-Install-Binary.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

GeorgeVrk, 2025-05-02:
Hello, I was able to follow the documentation clearly.
I downloaded ros2-kilted-beta-20250430-windows-release-amd64.zip from [https://github.com/ros2/ros2/releases/tag/release-kilted-beta-20250430](url) and followed the steps on my C drive with no issue. 

Specs:
OS Name:                   Microsoft Windows 10 Pro
OS Version:                10.0.19045 N/A Build 19045
System Type:               x64-based PC
Processor(s):              1 Processor(s) Installed.
Total Physical Memory:     32,688 MB

Result:
[image]

When I tried to reproduce this on my D drive, I could only run the demo_nodes_cpp talker & listener, while for the demo_nodes_py equivalent, I ran into the "failed to create process" error.
If pixi_ws was installed on my C drive, I could run both cpp and py nodes with no issue from my D drive as shown below : 
 
[image]

But if I deleted pixi_ws from my C drive I could no longer run the py nodes :

[image]

Not sure if I had to configure something, but thought I'd share nevertheless :)

mjcarroll, 2025-05-09:
> If pixi_ws was installed on my C drive, I could run both cpp and py nodes with no issue from my D drive as shown below :

This is a known limitation of the Windows binary builds at the moment. The `pixi_ws` must reside in the same place that it was when the binaries were built for python.  More discussion here: https://github.com/ros2/ros2/issues/1675

closed 2025-05-09 by mjcarroll