ros2/kilted_tutorial_party #545: Working with Eclipse CycloneDDS

tfoote, 2025-05-01:
Check the documentation for the 'Working with Eclipse CycloneDDS' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Working with Eclipse CycloneDDS page](https://docs.ros.org/en/kilted/Installation/RMW-Implementations/DDS-Implementations/Working-with-Eclipse-CycloneDDS.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

jncfa, 2025-05-02:
Tested with Ubuntu Docker Container for arm64 and amd64(under Rosetta 2) (ghcr.io/sloretz/ros-testing:kilted-desktop)

The documentation seems clear and I don't find any obvious errors.

Testing with the demo nodes seems to work fine:
[image]

[image]

But I can't seem to get a build from source following the documentation:
```
root@38c88b22d206:/workspaces/vscode_ros2_workspace# colcon build --symlink-install 
Starting >>> cyclonedds
--- stderr: cyclonedds
../../../bin/idlc: symbol lookup error: ../../../bin/idlc: undefined symbol: generate_type_meta_ser
gmake[2]:  [src/tools/ddsperf/CMakeFiles/ddsperf_types_generate.dir/build.make:76: src/tools/ddsperf/ddsperf_types.c] Error 127
gmake[1]:  [CMakeFiles/Makefile2:553: src/tools/ddsperf/CMakeFiles/ddsperf_types_generate.dir/all] Error 2
gmake: *** [Makefile:156: all] Error 2
---
Failed   <<< cyclonedds [15.9s, exited with code 2]
```

Related issue on cyclonedds issue tracker: https://github.com/eclipse-cyclonedds/cyclonedds/issues/332

mjcarroll, 2025-05-09:
> But I can't seem to get a build from source following the documentation:

Can you open an issue upstream on `rmw_cyclonedds` with this error please?

closed 2025-05-09 by mjcarroll