ros2/kilted_tutorial_party #546: Working with eProsima Fast DDS

tfoote, 2025-05-01:
Check the documentation for the 'Working with eProsima Fast DDS' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Working with eProsima Fast DDS page](https://docs.ros.org/en/kilted/Installation/RMW-Implementations/DDS-Implementations/Working-with-eProsima-Fast-DDS.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

jncfa, 2025-05-02:
I was able to follow the documentation, and do both a local build and run the demo nodes:

```
root@2834b2eaeaad:/workspaces/vscode_ros2_workspace# rosdep install --from src -i
executing command [apt-get install ros-kilted-osrf-testing-tools-cpp]
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following NEW packages will be installed:
  ros-kilted-osrf-testing-tools-cpp
0 upgraded, 1 newly installed, 0 to remove and 1 not upgraded.
Need to get 934 kB of archives.
After this operation, 1152 kB of additional disk space will be used.
Get:1 http://packages.ros.org/ros2-testing/ubuntu noble/main amd64 ros-kilted-osrf-testing-tools-cpp amd64 2.2.0-2noble.20250424.202735 [934 kB]
Fetched 934 kB in 1s (668 kB/s)                            
debconf: delaying package configuration, since apt-utils is not installed
Selecting previously unselected package ros-kilted-osrf-testing-tools-cpp.
(Reading database ... 115970 files and directories currently installed.)
Preparing to unpack .../ros-kilted-osrf-testing-tools-cpp_2.2.0-2noble.20250424.202735_amd64.deb ...
Unpacking ros-kilted-osrf-testing-tools-cpp (2.2.0-2noble.20250424.202735) ...
Setting up ros-kilted-osrf-testing-tools-cpp (2.2.0-2noble.20250424.202735) ...
executing command [apt-get install ros-kilted-test-msgs]
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following NEW packages will be installed:
  ros-kilted-test-msgs
0 upgraded, 1 newly installed, 0 to remove and 1 not upgraded.
Need to get 648 kB of archives.
After this operation, 10.4 MB of additional disk space will be used.
Get:1 http://packages.ros.org/ros2-testing/ubuntu noble/main amd64 ros-kilted-test-msgs amd64 2.3.0-2noble.20250424.210646 [648 kB]
Fetched 648 kB in 1s (527 kB/s)               
debconf: delaying package configuration, since apt-utils is not installed
Selecting previously unselected package ros-kilted-test-msgs.
(Reading database ... 116013 files and directories currently installed.)
Preparing to unpack .../ros-kilted-test-msgs_2.3.0-2noble.20250424.210646_amd64.deb ...
Unpacking ros-kilted-test-msgs (2.3.0-2noble.20250424.210646) ...
Setting up ros-kilted-test-msgs (2.3.0-2noble.20250424.210646) ...
executing command [apt-get install libasio-dev]
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following NEW packages will be installed:
  libasio-dev
0 upgraded, 1 newly installed, 0 to remove and 1 not upgraded.
Need to get 443 kB of archives.
After this operation, 6141 kB of additional disk space will be used.
Get:1 http://archive.ubuntu.com/ubuntu noble/universe amd64 libasio-dev all 1:1.28.1-0.2 [443 kB]
Fetched 443 kB in 0s (1085 kB/s)    
debconf: delaying package configuration, since apt-utils is not installed
Selecting previously unselected package libasio-dev.
(Reading database ... 116594 files and directories currently installed.)
Preparing to unpack .../libasio-dev_1%3a1.28.1-0.2_all.deb ...
Unpacking libasio-dev (1:1.28.1-0.2) ...
Setting up libasio-dev (1:1.28.1-0.2) ...
#All required rosdeps installed successfully
root@2834b2eaeaad:/workspaces/vscode_ros2_workspace# colcon build --symlink-install
Starting >>> fastdds 
[Processing: fastdds]                             
[Processing: fastdds]                                     
[Processing: fastdds]                                       
[Processing: fastdds]                                       
[Processing: fastdds]                                       
[Processing: fastdds]                                       
[Processing: fastdds]                                       
Finished <<< fastdds [3min 54s]                              
Starting >>> rmw_fastrtps_shared_cpp
[Processing: rmw_fastrtps_shared_cpp]                                  
Finished <<< rmw_fastrtps_shared_cpp [40.4s]                            
Starting >>> rmw_fastrtps_cpp
Starting >>> rmw_fastrtps_dynamic_cpp
[Processing: rmw_fastrtps_cpp, rmw_fastrtps_dynamic_cpp]                                                                 
Finished <<< rmw_fastrtps_cpp [51.1s]                                                                                     
Finished <<< rmw_fastrtps_dynamic_cpp [57.3s]                            

Summary: 4 packages finished [5min 31s]
root@2834b2eaeaad:/workspaces/vscode_ros2_workspace# 
```

SANJAY-16012, 2025-05-04:
### Checks

☑️ I was able to follow the documentation.
☑️ The documentation seemed clear to me.
☑️ The documentation didn't have any obvious errors.

### Setup
    Rmw: Fastdds
    BuildType: Debian
    Os: Noble
    Chip: Amd64

[image]

**Can be closed...**

ahcorde, 2025-05-05:
Thank you for testing