ros2/kilted_tutorial_party #547: Working with GurumNetworks GurumDDS

tfoote, 2025-05-01:
Check the documentation for the 'Working with GurumNetworks GurumDDS' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Working with GurumNetworks GurumDDS page](https://docs.ros.org/en/kilted/Installation/RMW-Implementations/DDS-Implementations/Working-with-GurumNetworks-GurumDDS.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

mjcarroll, 2025-05-02:
GurumDDS is not building for Kilted, please disregard this ticket: https://github.com/ros2/rmw_gurumdds/issues/16

closed 2025-05-02 by mjcarroll

jncfa, 2025-05-02:
it seems the builds are failing on the buildfarm?
https://build.ros2.org/job/Kbin_uN64__gurumdds_cmake_module__ubuntu_noble_amd64__binary/
https://build.ros2.org/job/Kbin_uN64__rmw_gurumdds_cpp__ubuntu_noble_amd64__binary/

from the last [build log](https://build.ros2.org/job/Kbin_uN64__gurumdds_cmake_module__ubuntu_noble_amd64__binary/12/console), this seems to be the reason why the builds are failing:
`KeyError: "No packages available for 'gurumdds-3.2'"`

I only see gurumdds-3.0 on rosdistro: https://github.com/search?q=repo%3Aros%2Frosdistro+gurumdds-3.0&type=code