ros2/kilted_tutorial_party #548: Working with RTI Connext DDS

tfoote, 2025-05-01:
Check the documentation for the 'Working with RTI Connext DDS' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Working with RTI Connext DDS page](https://docs.ros.org/en/kilted/Installation/RMW-Implementations/DDS-Implementations/Working-with-RTI-Connext-DDS.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

jncfa, 2025-05-02:
I can take this one

jncfa, 2025-05-02:
There is something wrong with the prebuilt binaries, it appears they do not include the shared library objects and thus can't be used. Not sure what changed with the packaging or if something went wrong on the build farm, but you can see that the deb package no longer provides those files:
```
root@3db77bd4d846:/workspaces/vscode_ros2_workspace/test# curl --output test.deb http://packages.ros.org/ros2-testing/ubuntu/pool/main/r/ros-kilted-rmw-connextdds/ros-kilted-rmw-connextdds_1.1.0-2noble.20250424.210815_amd64.deb 
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  7176  100  7176    0     0  25473      0 --:--:-- --:--:-- --:--:-- 25906
root@3db77bd4d846:/workspaces/vscode_ros2_workspace/test# dpkg-deb -R test.deb test
root@3db77bd4d846:/workspaces/vscode_ros2_workspace/test# find test -name *lib*
root@3db77bd4d846:/workspaces/vscode_ros2_workspace/test# curl --output test-right.deb http://packages.ros.org/ros2-testing/ubuntu/pool/main/r/ros-humble-rmw-connextdds/ros-humble-rmw-connextdds_0.11.3-1jammy.20250325.191149_amd64.deb
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 16360  100 16360    0     0  40620      0 --:--:-- --:--:-- --:--:-- 41105
root@3db77bd4d846:/workspaces/vscode_ros2_workspace/test# dpkg-deb -R test-right.deb test-right
root@3db77bd4d846:/workspaces/vscode_ros2_workspace/test# find test-right -name *lib*
test-right/opt/ros/humble/lib
test-right/opt/ros/humble/lib/librmw_connextdds.so
test-right/opt/ros/humble/share/rmw_connextdds/cmake/ament_cmake_export_libraries-extras.cmake
test-right/opt/ros/humble/share/rmw_connextdds/environment/library_path.dsv
test-right/opt/ros/humble/share/rmw_connextdds/environment/library_path.sh
```

jncfa, 2025-05-02:
If I build and source locally rmw_connextdds, it will work just fine:

[image]

[image]

jncfa, 2025-05-02:
I would leave two notes about the documentation however:
- Adding `rosdep install --from src -i`, to both put it more in line with the other [rmw docs](https://docs.ros.org/en/kilted/Installation/RMW-Implementations/DDS-Implementations/Working-with-eProsima-Fast-DDS.html#build-from-source-code)
- Suggest `source $(find /opt -name rtisetenv*.bash)` to automatically source variables needed

I can make a PR if these suggestions seem OK with you

fujitatomoya, 2025-05-02:
see https://github.com/ros2/kilted_tutorial_party/issues/282

fujitatomoya, 2025-05-05:
@jncfa we have released new package and binaries, can you recheck this?

jncfa, 2025-05-05:
> [@jncfa](https://github.com/jncfa) we have released new package and binaries, can you recheck this?

yep, new package is working fine!

[image]

regarding my suggestions, do you see value in adding these changes or something like it on the documentation page?

> I would leave two notes about the documentation however:
>     - Adding `rosdep install --from src -i`, to both put it more in line with the other [rmw [...]

fujitatomoya, 2025-05-06:
> - Adding rosdep install --from src -i, to both put it more in line with the other [rmw docs](https://docs.ros.org/en/kilted/Installation/RMW-Implementations/DDS-Implementations/Working-with-eProsima-Fast-DDS.html#build-from-source-code)

this requires to be called before doing the source build.

> - Suggest source $(find /opt -name rtisetenv*.bash) to automatically source variables needed

yeah i was aware of this a few days ago, and dropped it... thanks for reminding me of this.
maybe we can specify the full path for each version, instead it would source multiple files and could be messed up?

after all, both makes sense to me. please open the PR, i am happy to review!

closed 2025-05-09 by mjcarroll