ros2/kilted_tutorial_party #549: Working with Zenoh

tfoote, 2025-05-01:
Check the documentation for the 'Working with Zenoh' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Working with Zenoh page](https://docs.ros.org/en/kilted/Installation/RMW-Implementations/Non-DDS-Implementations/Working-with-Zenoh.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

khaledgabr77, 2025-05-01:
i'm testing this one

khaledgabr77, 2025-05-01:
There is an issue in the documentation [here](https://docs.ros.org/en/kilted/Installation/RMW-Implementations/Non-DDS-Implementations/Working-with-Zenoh.html#:~:text=ros2%20run%20rmw_zenoh_cpp%20talker%20rmw_zenohd)

The command shown:

`ros2 run rmw_zenoh_cpp talker rmw_zenohd
`

is incorrect. It should be:

`ros2 run rmw_zenoh_cpp  rmw_zenohd`

danielcranston, 2025-05-03:
Indeed 👍 It's being addressed in https://github.com/ros2/ros2_documentation/pull/5465

closed 2025-05-05 by ahcorde