ros2/kilted_tutorial_party #550: Writing a Composable Node

tfoote, 2025-05-01:
Check the documentation for the 'Writing a Composable Node' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Writing a Composable Node page](https://docs.ros.org/en/kilted/Tutorials/Intermediate/Writing-a-Composable-Node.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

avanmalleghem, 2025-05-02:
Interested in

avanmalleghem, 2025-05-04:
Ok once PR merged : https://github.com/ros2/ros2_documentation/pull/5529

christophebedard, 2025-05-04:
Thanks for testing.

closed 2025-05-04 by christophebedard