ros2/kilted_tutorial_party #551: Writing A Simple Cpp Publisher And Subscriber

tfoote, 2025-05-01:
Check the documentation for the 'Writing A Simple Cpp Publisher And Subscriber' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Writing A Simple Cpp Publisher And Subscriber page](https://docs.ros.org/en/kilted/Tutorials/Beginner-Client-Libraries/Writing-A-Simple-Cpp-Publisher-And-Subscriber.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

dheerubhai-101, 2025-05-01:
- [x] I was able to follow the documentation.
- [x] The documentation seemed clear to me.
- [ ] The documentation didn't have any obvious errors.

Similar issues as in https://github.com/ros2/kilted_tutorial_party/issues/552. Colcon build does not work with target_link in CMakeLists.txt, and using ament_target_dependencies shows deprecation warning. Will this need some environment declaration for `${std_msgs_TARGETS}`?

```
vscode ➜ /workspaces/kilted_ws $ colcon build --packages-select cpp_pubsub
Starting >>> cpp_pubsub
--- stderr: cpp_pubsub                             
/workspaces/kilted_ws/src/cpp_pubsub/src/subscriber_lambda_function.cpp:17:10: fatal error: rclcpp/rclcpp.hpp: No such file or directory
   17 | #include "rclcpp/rclcpp.hpp"
      |          ^~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/listener.dir/build.make:76: CMakeFiles/listener.dir/src/subscriber_lambda_function.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:165: CMakeFiles/listener.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< cpp_pubsub [3.04s, exited with code 2]

Summary: 0 packages finished [3.14s]
  1 package failed: cpp_pubsub
  1 package had stderr output: cpp_pubsub
```

```
vscode ➜ /workspaces/kilted_ws $ colcon build --packages-select cpp_pubsub
Starting >>> cpp_pubsub
--- stderr: cpp_pubsub                         
CMake Deprecation Warning at /opt/ros/kilted/share/ament_cmake_target_dependencies/cmake/ament_target_dependencies.cmake:87 (message):
  ament_target_dependencies() is deprecated.  Use target_link_libraries()
  with modern CMake targets instead.  Try replacing this call with:

    target_link_libraries(talker PUBLIC
      ${std_msgs_TARGETS}
      rclcpp::rclcpp
    )

Call Stack (most recent call first):
  CMakeLists.txt:18 (ament_target_dependencies)

CMake Error at CMakeLists.txt:20 (install):
  install TARGETS given target "listener" which does not exist.

gmake: *** [Makefile:267: cmake_check_build_system] Error 1
---
Failed   <<< cpp_pubsub [0.41s, exited with code 2]

Summary: 0 packages finished [0.50s]
  1 package failed: cpp_pubsub
  1 package had stderr output: cpp_pubsub
vscode ➜ /workspaces/kilted_ws $ colcon build --packages-select cpp_pubsub
Starting >>> cpp_pubsub
--- stderr: cpp_pubsub                              
CMake Deprecation Warning at /opt/ros/kilted/share/ament_cmake_target_dependencies/cmake/ament_target_dependencies.cmake:87 (message):
  ament_target_dependencies() is deprecated.  Use target_link_libraries()
  with modern CMake targets instead.  Try replacing this call with:

    target_link_libraries(talker PUBLIC
      ${std_msgs_TARGETS}
      rclcpp::rclcpp
    )

Call Stack (most recent call first):
  CMakeLists.txt:18 (ament_target_dependencies)

CMake Deprecation Warning at /opt/ros/kilted/share/ament_cmake_target_dependencies/cmake/ament_target_dependencies.cmake:87 (message):
  ament_target_dependencies() is deprecated.  Use target_link_libraries()
  with modern CMake targets instead.  Try replacing this call with:

    target_link_libraries(listener PUBLIC
      ${std_msgs_TARGETS}
      rclcpp::rclcpp
    )

Call Stack (most recent call first):
  CMakeLists.txt:21 (ament_target_dependencies)

---
Finished <<< cpp_pubsub [4.89s]

Summary: 1 package finished [4.98s]
  1 package had stderr output: cpp_pubsub
```

christophebedard, 2025-05-03:
Thanks for testing! I was able to reproduce the issue on my end too.

This part: https://docs.ros.org/en/kilted/Tutorials/Beginner-Client-Libraries/Writing-A-Simple-Cpp-Publisher-And-Subscriber.html#id2

```cmake
  add_executable(listener src/subscriber_lambda_function.cpp)
  target_link_libraries(talker PUBLIC rclcpp::rclcpp ${std_msgs_TARGETS})
```

should be this:

```cmake
  add_executable(listener src/subscriber_lambda_function.cpp)
  target_link_libraries(listener PUBLIC rclcpp::rclcpp ${std_msgs_TARGETS})
```

Note that `target_link_libraries` uses the `listener` target and not `talker`, which explains why `rclcpp` couldn't be found. I opened https://github.com/ros2/ros2_documentation/pull/5481 to fix the docs. It would be helpful if you could test with that fix and let us know if it works for you!

dheerubhai-101, 2025-05-03:
and yes, it does build and work now after the correct target in CMakeLists.txt

christophebedard, 2025-05-03:
Thanks for confirming! And don't worry, this kind of thing is so easy to miss 😁

closed 2025-05-03 by christophebedard