ros2/kilted_tutorial_party #552: Writing A Simple Cpp Service And Client

tfoote, 2025-05-01:
Check the documentation for the 'Writing A Simple Cpp Service And Client' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Writing A Simple Cpp Service And Client page](https://docs.ros.org/en/kilted/Tutorials/Beginner-Client-Libraries/Writing-A-Simple-Cpp-Service-And-Client.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

dheerubhai-101, 2025-05-01:
hi, testing this

dheerubhai-101, 2025-05-01:
- [x] I was able to follow the documentation.
- [x] The documentation seemed clear to me.
- [ ] The documentation didn't have any obvious errors.

In `CMakeLists.txt`, `target_link_libraries(server PUBLIC rclcpp::rclcpp ${example_interfaces_TARGETS})` seems to cause build issues. 
```
vscode ➜ /workspaces/kilted_ws $ colcon build --packages-select cpp_srvcli
Starting >>> cpp_srvcli
--- stderr: cpp_srvcli
/workspaces/kilted_ws/src/cpp_srvcli/src/add_two_ints_client.cpp:5:10: fatal error: example_interfaces/srv/add_two_ints.hpp: No such file or directory
    5 | #include "example_interfaces/srv/add_two_ints.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/client.dir/build.make:76: CMakeFiles/client.dir/src/add_two_ints_client.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:165: CMakeFiles/client.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< cpp_srvcli [0.08s, exited with code 2]

Summary: 0 packages finished [0.18s]
  1 package failed: cpp_srvcli
  1 package had stderr output: cpp_srvcli
```
When using `ament_target_dependencies(server rclcpp example_interfaces)` instead, it builds but with deprecation warnings. 
```
vscode ➜ /workspaces/kilted_ws $ colcon build --packages-select cpp_srvcli
Starting >>> cpp_srvcli
--- stderr: cpp_srvcli                              
CMake Deprecation Warning at /opt/ros/kilted/share/ament_cmake_target_dependencies/cmake/ament_target_dependencies.cmake:87 (message):
  ament_target_dependencies() is deprecated.  Use target_link_libraries()
  with modern CMake targets instead.  Try replacing this call with:

    target_link_libraries(server PUBLIC
      ${example_interfaces_TARGETS}
      rclcpp::rclcpp
    )

Call Stack (most recent call first):
  CMakeLists.txt:9 (ament_target_dependencies)

CMake Deprecation Warning at /opt/ros/kilted/share/ament_cmake_target_dependencies/cmake/ament_target_dependencies.cmake:87 (message):
  ament_target_dependencies() is deprecated.  Use target_link_libraries()
  with modern CMake targets instead.  Try replacing this call with:

    target_link_libraries(client PUBLIC
      ${example_interfaces_TARGETS}
      rclcpp::rclcpp
    )

Call Stack (most recent call first):
  CMakeLists.txt:12 (ament_target_dependencies)

---
Finished <<< cpp_srvcli [2.74s]

Summary: 1 package finished [2.83s]
  1 package had stderr output: cpp_srvcli
```

Also small issues with docs page. Section 2.2 Add executable has a small issue with the `copy` command. It adds new lines on pasting. Looks like the same is the issue with older distro documentations as well.

[image]

 
Section 3.2 has similar webpage issue with `copy` command

christophebedard, 2025-05-03:
Thanks for testing and for providing info about what doesn't work!

As mentioned in https://github.com/ros2/kilted_tutorial_party/issues/551#issuecomment-2848371983, the fix for the build is https://github.com/ros2/ros2_documentation/pull/5481.

As for the copy button, this should fix it: https://github.com/ros2/ros2_documentation/pull/5480.

closed 2025-05-03 by christophebedard

dheerubhai-101, 2025-05-03:
Hi Chris, yes just a small fix in the Ros Docs as you said, https://github.com/ros2/ros2_documentation/pull/5486 should fix it