ros2/kilted_tutorial_party #553: Writing A Simple Py Publisher And Subscriber

tfoote, 2025-05-01:
Check the documentation for the 'Writing A Simple Py Publisher And Subscriber' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Writing A Simple Py Publisher And Subscriber page](https://docs.ros.org/en/kilted/Tutorials/Beginner-Client-Libraries/Writing-A-Simple-Py-Publisher-And-Subscriber.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

dheerubhai-101, 2025-05-01:
hi, I'll be trying this

dheerubhai-101, 2025-05-01:
- [x] I was able to follow the documentation.
- [x] The documentation seemed clear to me.
- [x] The documentation didn't have any obvious errors.

[image]

ahcorde, 2025-05-01:
Thank you for testing @dheerubhai-101

closed 2025-05-01 by ahcorde