ros2/kilted_tutorial_party #554: Writing A Simple Py Service And Client

tfoote, 2025-05-01:
Check the documentation for the 'Writing A Simple Py Service And Client' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Writing A Simple Py Service And Client page](https://docs.ros.org/en/kilted/Tutorials/Beginner-Client-Libraries/Writing-A-Simple-Py-Service-And-Client.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

dheerubhai-101, 2025-05-01:
hi, I'll be trying this

christophebedard, 2025-05-01:
Thanks for testing!

closed 2025-05-01 by christophebedard