ros2/kilted_tutorial_party #556: Writing A Tf2 Broadcaster Py

tfoote, 2025-05-01:
Check the documentation for the 'Writing A Tf2 Broadcaster Py' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Writing A Tf2 Broadcaster Py page](https://docs.ros.org/en/kilted/Tutorials/Intermediate/Tf2/Writing-A-Tf2-Broadcaster-Py.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

akky20, 2025-05-01:
Interested

Aniketkumarroy, 2025-05-04:
Hi I tested this feature and found that everything is alright except that now the turtlesim message is contained in `turtlesim_msgs` instead of `turtlesim`. but in both code given and downloaded from wget uses `turtlesim` which throws error.
by changing the line `from turtlesim.msg import Pose` to `from turtlesim_msgs.msg import Pose` it works fine

I have addressed this issue in this PR https://github.com/ros2/ros2_documentation/pull/5526
here are the results

[image]

- [x] I was able to follow the documentation.
- [x] The documentation seemed clear to me.
- [ ] The documentation didn't have any obvious errors.

christophebedard, 2025-05-04:
Thanks for testing. Since a PR was created to fix the docs, I'll close this.

closed 2025-05-04 by christophebedard