ros2/kilted_tutorial_party #557: Writing A Tf2 Listener Cpp

tfoote, 2025-05-01:
Check the documentation for the 'Writing A Tf2 Listener Cpp' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Writing A Tf2 Listener Cpp page](https://docs.ros.org/en/kilted/Tutorials/Intermediate/Tf2/Writing-A-Tf2-Listener-Cpp.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

ahcorde, 2025-05-02:
Do you mind to open a PR in ros2_documentation with the mentioned modification?

danielcranston, 2025-05-06:
@jmackay2 a kilted branch has been made in geometry_tutorials now, fyi.

ahcorde, 2025-05-08:
Thank you for testing

closed 2025-05-08 by ahcorde