ros2/kilted_tutorial_party #558: Writing A Tf2 Listener Py

tfoote, 2025-05-01:
Check the documentation for the 'Writing A Tf2 Listener Py' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Writing A Tf2 Listener Py page](https://docs.ros.org/en/kilted/Tutorials/Intermediate/Tf2/Writing-A-Tf2-Listener-Py.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

akky20, 2025-05-01:
Interested

christophebedard, 2025-05-01:
Please pick a single issue, validate it, and report your findings. Then you can pick another issue.

Aniketkumarroy, 2025-05-04:
Hi I tested this feature and found that everything is alright except that now the turtlesim services is contained in `turtlesim_msgs` instead of `turtlesim`. but the code both on site and downloaded from wget uses `turtlesim` which throws error.
by changing the line `from turtlesim.srv import Spawn` to `from turtlesim_msgs.srv import Spawn` it works fine

I have addressed this issue in this PR https://github.com/ros2/ros2_documentation/pull/5526
here are the results

[image]

- [x] I was able to follow the documentation.
- [x] The documentation seemed clear to me.
- [ ] The documentation didn't have any obvious errors.

christophebedard, 2025-05-04:
Thanks for testing. Since a PR was created to fix the docs, I'll close this.

closed 2025-05-04 by christophebedard