ros2/kilted_tutorial_party #559: Writing A Tf2 Static Broadcaster Cpp

tfoote, 2025-05-01:
Check the documentation for the 'Writing A Tf2 Static Broadcaster Cpp' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Writing A Tf2 Static Broadcaster Cpp page](https://docs.ros.org/en/kilted/Tutorials/Intermediate/Tf2/Writing-A-Tf2-Static-Broadcaster-Cpp.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

mayankysharma, 2025-05-01:
Hi ,
I am testing this.

virkhanna20, 2025-05-04:
I am working on this for the past 2 days.

SANJAY-16012, 2025-05-04:
### **Checks**

☑️I was able to follow the documentation.
☑️The documentation seemed clear to me.
☑️The documentation didn't have any obvious errors.

[image]

[image]

[image]

[image]

ahcorde, 2025-05-05:
Thank you for testing

closed 2025-05-05 by ahcorde