ros2/kilted_tutorial_party #56: Check ROS 2 Multicast CLI

tfoote, 2025-05-01:
## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Checks
- [ ] **Check send and receive**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 multicast receive
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 multicast send
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 1
       Waiting for UDP multicast datagram...
       Received from 10.211.55.9:59527: 'Hello World!'
       ```
    1.
       ```bash
       # stdout in terminal 2
       Sending one UDP multicast datagram...
       ```
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

rtarun1, 2025-05-02:
## Setup 

- Rmw: Fastdds
- BuildType: Debian
- Chip: Amd64
- Inside Docker container running `ghcr.io/sloretz/ros-testing:kilted-desktop`

## Output 

- [x] Check send and receive

[image]

ahcorde, 2025-05-02:
Thank you for testing

closed 2025-05-02 by ahcorde