ros2/kilted_tutorial_party #560: Writing A Tf2 Static Broadcaster Py

tfoote, 2025-05-01:
Check the documentation for the 'Writing A Tf2 Static Broadcaster Py' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Writing A Tf2 Static Broadcaster Py page](https://docs.ros.org/en/kilted/Tutorials/Intermediate/Tf2/Writing-A-Tf2-Static-Broadcaster-Py.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

akky20, 2025-05-01:
Interested

christophebedard, 2025-05-01:
Please make sure to pick a single issue, validate it, and report your findings. Then you can pick another issue.

Sanoronas, 2025-05-02:
I'm also interested, in case @akky20 is assigned to something else

akky20, 2025-05-02:
Working fine !! followed the mentioned tutorial

[image]

akky20, 2025-05-02:
@Sanoronas Sorry mate, already completed just positng the results was left

christophebedard, 2025-05-02:
Thanks for testing

closed 2025-05-02 by christophebedard

christophebedard, 2025-05-02:
Can you be more specific?

1. What are the discrepancies?
2. What are the duplicate explanations?
1. What commands are not intuitive/what are the placeholders?

Sanoronas, 2025-05-03:
@christophebedard 

> 1. What are the discrepancies?

There is a try-except block in main function of example code which isn't there in the downloaded file. No functional issues with that. Just a difference.

> 3. What are the duplicate explanations?

This is the section I am refering to. Upon closer look it actually is two different cases described with very similar wording.

[image]

> 5. What commands are not intuitive/what are the placeholders?

Again refering to the screenshotted section. The terminal commands to start a static transform publisher contain placeholder values for x y z so when copy pasting it in terminal it gives an error. In my opinion it would be better to explain the syntax and then have a runnable commands as example.

christophebedard, 2025-05-03:
Thanks for the info, that really helps!

> There is a try-except block in main function of example code which isn't there in the downloaded file. No functional issues with that. Just a difference.

Looks like the links to the file need to be updated to point to the `rolling` branch instead of the `ros2` branch. I've opened a PR for that: https://github.com/ros2/ros2_documentation/pull/5505

> This is the section I am refering to. Upon closer look it actually is two different cases described with very similar wording.

Ah yeah, those do look similar. A PR that changes the second paragraph so it doesn't look too much like the first one would be welcome. The C++ version of the tutorial has the same section, so that could be changed too: https://docs.ros.org/en/kilted/Tutorials/Intermediate/Tf2/Writing-A-Tf2-Static-Broadcaster-Cpp.html#the-proper-way-to-publish-static-transforms.

> Again refering to the screenshotted section. The terminal commands to start a static transform publisher contain placeholder values for x y z so when copy pasting it in terminal it gives an error. In my opinion it would be better to explain the syntax and then have a runnable commands as example.

Yeah, I think that would be useful, similar to how the Python launch file example below uses actual values and not placeholders. A PR would be welcome.

akky20, 2025-05-04:
@Sanoronas Great, I also missed to notice this. Thanks !!