ros2/kilted_tutorial_party #561: Writing an Action Server and Client (Cpp)

tfoote, 2025-05-01:
Check the documentation for the 'Writing an action server and client' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Writing an action server and client (C++)](https://docs.ros.org/en/kilted/Tutorials/Intermediate/Writing-an-Action-Server-Client/Cpp.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

Tanishq30052002, 2025-05-01:
I will test this functionality

Tanishq30052002, 2025-05-02:
Following the documentation throws the following error:

[image]

Please Note: I have installed or modified anything other than what is asked in the documentation.
Also I have used the docker image

Tanishq30052002, 2025-05-02:
While creating a pkg using the cmd in the documentation:
```
cd ~/ros2_ws/src
ros2 pkg create --dependencies custom_action_interfaces rclcpp rclcpp_action rclcpp_components --license Apache-2.0 -- custom_action_cpp
```

custom_action_interfaces becomes a dependency, which is not installed in the ROS2 Kilted.

Tanishq30052002, 2025-05-02:
i am using the docker image mentioned on discourse

ahcorde, 2025-05-02:
As the tutorial said:

```
You will need the custom_action_interfaces package and the Fibonacci.action interface defined in the previous tutorial, [Creating an action](https://docs.ros.org/en/kilted/Tutorials/Intermediate/Creating-an-Action.html).
```

You might need to follow the previous tutorial to create this package

Tanishq30052002, 2025-05-02:
- I was able to follow the documentation. 
- The documentation seemed clear to me.
- The documentation didn't have any obvious errors.

[image]

ahcorde, 2025-05-02:
Thank you for testing

closed 2025-05-02 by ahcorde