ros2/kilted_tutorial_party #562: Writing an Action Server and Client (Python)

tfoote, 2025-05-01:
Check the documentation for the 'Writing an action server and client' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Writingan action server and client (Python)](https://docs.ros.org/en/kilted/Tutorials/Intermediate/Writing-an-Action-Server-Client/Py.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

akky20, 2025-05-01:
Hello I will try this

christophebedard, 2025-05-01:
Please make sure to pick a single issue, validate it, and report your findings. Then you can pick another issue.

akky20, 2025-05-02:
@christophebedard will do 
Thanks

ahcorde, 2025-05-02:
Thank you for testing

closed 2025-05-02 by ahcorde