ros2/kilted_tutorial_party #58: Check ROS 2 Node CLI

tfoote, 2025-05-01:
## Setup
- Rmw: Cyclone
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check info**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       Launch a node
       ```bash
       # User input in terminal 1
       ros2 launch demo_nodes_cpp talker_listener_launch.py
       ```
    1.
       Get the info for the `talker` that was launched.
       ```bash
       # User input in terminal 2
       ros2 node info /talker
       ```
  
    **Expect**
    
    1.
       You should see something similar to the following
       ```bash
       # stdout in terminal 2
       Subscribers:
         /parameter_events: rcl_interfaces/msg/ParameterEvent
       Publishers:
         /chatter: std_msgs/msg/String
         /parameter_events: rcl_interfaces/msg/ParameterEvent
         /rosout: rcl_interfaces/msg/Log
       Service Servers:
         /talker/describe_parameters: rcl_interfaces/srv/DescribeParameters
         /talker/get_parameter_types: rcl_interfaces/srv/GetParameterTypes
         /talker/get_parameters: rcl_interfaces/srv/GetParameters
         /talker/get_type_description: type_description_interfaces/srv/GetTypeDescription
         /talker/list_parameters: rcl_interfaces/srv/ListParameters
         /talker/set_parameters: rcl_interfaces/srv/SetParameters
         /talker/set_parameters_atomically: rcl_interfaces/srv/SetParametersAtomically
       Service Clients:
       
       Action Servers:
       
       Action Clients:
       ```
  </details>
- [ ] **Check list**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       Launch a node
       ```bash
       # User input in terminal 1
       ros2 launch demo_nodes_cpp talker_listener_launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 node list
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 2
       /listener
       /talker
       ```
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

ShahazadAbdulla, 2025-05-03:
I'll test this on Windows 11 / amd64 / binary (Pixi) / CycloneDDS.

ShahazadAbdulla, 2025-05-03:
Completed checks for Issue #58 on Windows 11 / amd64 / binary / CycloneDDS.

- **Check 1 (`ros2 node info /talker`):** The command executed successfully and the output structure was generally as expected. However, there was one difference: `/parameter_events: rcl_interfaces/msg/ParameterEvent` was listed under **Publishers only**, not under Subscribers as shown in the 'Expect' section of this issue. See actual output below.
- **Check 2 (`ros2 node list`):** Passed. The output correctly listed `/listener` and `/talker`.

**Actual Output for `ros2 node info /talker`:**

```text
(pixi_ros2_rolling) C:\pixi_ws>ros2 node info /talker
/talker
  Subscribers:

  Publishers:
    /chatter: std_msgs/msg/String
    /parameter_events: rcl_interfaces/msg/ParameterEvent
    /rosout: rcl_interfaces/msg/Log
  Service Servers:
    /talker/describe_parameters: rcl_interfaces/srv/DescribeParameters
    /talker/get_parameter_types: rcl_interfaces/srv/GetParameterTypes
    /talker/get_parameters: rcl_interfaces/srv/GetParameters
    /talker/get_type_description: type_description_interfaces/srv/GetTypeDescription
    /talker/list_parameters: rcl_interfaces/srv/ListParameters
    /talker/set_parameters: rcl_interfaces/srv/SetParameters
    /talker/set_parameters_atomically: rcl_interfaces/srv/SetParametersAtomically
  Service Clients:

  Action Servers:

  Action Clients:
```

Overall, the functionality seems to work, but the `node info` output has this slight deviation regarding the `/parameter_events` subscriber.

mjcarroll, 2025-05-09:
https://github.com/osrf/ros2_test_cases/issues/1695

closed 2025-05-09 by mjcarroll