ros2/kilted_tutorial_party #6: Allocator Template Tutorial

tfoote, 2025-05-01:
Check the documentation for the 'Allocator Template Tutorial' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Allocator Template Tutorial page](https://docs.ros.org/en/kilted/Tutorials/Advanced/Allocator-Template-Tutorial.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

Juliaj, 2025-05-02:
Followed the instructions from [Allocator Template Tutorial page](https://docs.ros.org/en/kilted/Tutorials/Advanced/Allocator-Template-Tutorial.html) and was able to create a custom memory allocator. 
 
- [x] I was able to follow the documentation.
- [x] The documentation seemed clear to me.
- [x] The documentation didn't have any obvious errors.

[image]

ahcorde, 2025-05-05:
Thank you for testing!

closed 2025-05-05 by ahcorde