ros2/kilted_tutorial_party #64: Check ROS 2 Param CLI

tfoote, 2025-05-01:
## Setup
- Rmw: Cyclone
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Checks
- [ ] **Check delete**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run turtlesim turtlesim_node
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 param delete /turtlesim background_b
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 2
       Deleting parameter failed: cannot undeclare a statically typed parameter
       ```
  </details>
- [ ] **Check describe**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run turtlesim turtlesim_node
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 param describe /turtlesim background_b
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 2
       Parameter name: background_b
       Type: integer
       Description: Blue channel of the background color
       Constraints:
         Min value: 0
         Max value: 255
         Step: 1
       ```
  </details>
- [ ] **Check dump**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run turtlesim turtlesim_node
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 param dump /turtlesim
       ```
  
    **Expect**
    
    1.
       Something like the following.
       ```bash
       # stdout in terminal 2
       /turtlesim:
         ros__parameters:
           background_b: 255
           background_g: 86
           background_r: 69
           holonomic: false
           qos_overrides:
             /parameter_events:
               publisher:
                 depth: 1000
                 durability: volatile
                 history: keep_last
                 reliability: reliable
           start_type_description_service: true
           use_sim_time: false
       ```
  </details>
- [ ] **Check get**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run turtlesim turtlesim_node
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 param get /turtlesim background_b
       ```
  
    **Expect**
    
    1.
       Something like the following.
       ```bash
       # stdout in terminal 2
       Integer value is: 255
       ```
  </details>
- [ ] **Check list**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run turtlesim turtlesim_node
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 param list /turtlesim
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 2
       ros2 param list /turtlesim
       background_b
       background_g
       background_r
       holonomic
       qos_overrides./parameter_events.publisher.depth
       qos_overrides./parameter_events.publisher.durability
       qos_overrides./parameter_events.publisher.history
       qos_overrides./parameter_events.publisher.reliability
       start_type_description_service
       use_sim_time
       ```
  </details>
- [ ] **Check set**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run turtlesim turtlesim_node
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 param set /turtlesim background_b 0
       ```
  
    **Expect**
    
    1.
       The background in the turtlesim window should have become green
       ```bash
       # stdout in terminal 2
       Set parameter successful
       ```
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

suchetanrs, 2025-05-02:
Working on this

suchetanrs, 2025-05-02:
All of them work fine.

All outputs are below:

```
root@cluttered:~# source /opt/ros/kilted/setup.bash
root@cluttered:~# export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
root@cluttered:~# ./scripts/ros2_param_cli.sh 
[INFO] [1746216698.071791002] [turtlesim]: Starting turtlesim with node name /turtlesim
[INFO] [1746216698.080166922] [turtlesim]: Spawning turtle [turtle1] at x=[5.544445], y=[5.544445], theta=[0.000000]
glx: failed to create dri3 screen
failed to load driver: nouveau
glx: failed to create dri3 screen
failed to load driver: nouveau
==========================ros2 param delete /turtlesim background_b
Deleting parameter failed: cannot undeclare a statically typed parameter
==========================ros2 param describe /turtlesim background_b
Parameter name: background_b
  Type: integer
  Description: Blue channel of the background color
  Constraints:
    Min value: 0
    Max value: 255
    Step: 1
==========================ros2 param dump /turtlesim
/turtlesim:
  ros__parameters:
    background_b: 255
    background_g: 86
    background_r: 69
    holonomic: false
    qos_overrides:
      /parameter_events:
        publisher:
          depth: 1000
          durability: volatile
          history: keep_last
          reliability: reliable
    start_type_description_service: true
    use_sim_time: false

==========================ros2 param get /turtlesim background_b
Integer value is: 255
==========================ros2 param list /turtlesim
  background_b
  background_g
  background_r
  holonomic
  qos_overrides./parameter_events.publisher.depth
  qos_overrides./parameter_events.publisher.durability
  qos_overrides./parameter_events.publisher.history
  qos_overrides./parameter_events.publisher.reliability
  start_type_description_service
  use_sim_time
==========================ros2 param set /turtlesim background_b 255
Set parameter successful
root@cluttered:~# 

```

closed 2025-05-02 by ahcorde