ros2/kilted_tutorial_party #68: Check ROS 2 Pkg CLI

tfoote, 2025-05-01:
## Setup
- Rmw: Cyclone
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check create**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 pkg create my-pkg
       ```
  
    **Expect**
    
    1.
       Creates a folder named `my-pkg`, which has a `package.xml` file, `CMakeLists.txt` file, `src` folder, and `include` folder.
    
  </details>
- [ ] **Check executables**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 pkg executables turtlesim
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 1
       turtlesim draw_square
       turtlesim mimic
       turtlesim turtle_teleop_key
       turtlesim turtlesim_node
       ```
  </details>
- [ ] **Check list**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 pkg list
       ```
  
    **Expect**
    
    1.
       Lists all the packages, for example.
       ```bash
       # stdout in terminal 1
       action_msgs
       action_tutorials_cpp
       action_tutorials_interfaces
       ...
       visualization_msgs
       yaml_cpp_vendor
       zstd_vendor
       ```
  </details>
- [ ] **Check prefix**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 pkg prefix turtlesim
       ```
  
    **Expect**
    
    1.
       This varies depending on your install type. For example, from binaries on Ubuntu, it will be `/opt/ros/iron`
    
  </details>
- [ ] **Check xml**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 pkg xml turtlesim
       ```
  
    **Expect**
    
    1.
       The full `package.xml` of the `turtlesim` package is printed to stdout.
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

ShahazadAbdulla, 2025-05-03:
I'll test this on Windows 11 / amd64 / binary / CycloneDDS.

ShahazadAbdulla, 2025-05-03:
Successfully completed all checks for Issue #68 on Windows 11 / amd64 / binary installation using **CycloneDDS** (`rmw_cyclonedds_cpp`).

All `ros2 pkg` commands executed successfully and the outputs matched the expectations, identical to the Fast DDS test.

ahcorde, 2025-05-05:
Thank you for testing

closed 2025-05-05 by ahcorde