ros2/kilted_tutorial_party #7: Alternatives

tfoote, 2025-05-01:
Check the documentation for the 'Alternatives' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Alternatives page](https://docs.ros.org/en/kilted/Installation/Alternatives.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

fujitatomoya, 2025-05-02:
- https://docs.ros.org/en/kilted/Installation/Alternatives/Ubuntu-Install-Binary.html

  works without any problems.

```console
root@tomoyafujita:~# . ~/ros2_kilted_binary/ros2-linux/setup.bash
root@tomoyafujita:~# ros2 run demo_nodes_cpp talker
[INFO] [1746163357.374508775] [talker]: Publishing: 'Hello World: 1'
[INFO] [1746163358.374436941] [talker]: Publishing: 'Hello World: 2'
[INFO] [1746163359.374437437] [talker]: Publishing: 'Hello World: 3'
[INFO] [1746163360.374436527] [talker]: Publishing: 'Hello World: 4'
^C[ros2run]: Received signal:  Interrupt
[INFO] [1746163360.844088802] [rclcpp]: signal_handler(signum=2)
[INFO] [1746163360.844253268] [rclcpp]: signal_handler(signum=2)
```

fujitatomoya, 2025-05-02:
- https://docs.ros.org/en/kilted/Installation/Alternatives/Ubuntu-Development-Setup.html

  source build also works without any problems.

```console
...
---
Finished <<< rqt_bag_plugins [4.11s]
[Processing: rviz_default_plugins]
[Processing: rviz_default_plugins]
[Processing: rviz_default_plugins]
[Processing: rviz_default_plugins]
[Processing: rviz_default_plugins]
[Processing: rviz_default_plugins]
[Processing: rviz_default_plugins]
Finished <<< rviz_default_plugins [5min 23s]
Starting >>> rviz2
Finished <<< rviz2 [7.41s]

Summary: 370 packages finished [26min 55s]
  110 packages had stderr output: action_tutorials_py ament_clang_format ament_clang_tidy ament_copyright ament_cppcheck ament_cpplint ament_flake8 ament_index_python ament_lint ament_lint_cmake ament_mypy ament_package ament_pclint ament_pep257 ament_pycodestyle ament_pyflakes ament_uncrustify ament_xmllint camera_info_manager_py demo_nodes_py domain_coordinator examples_rclpy_executors examples_rclpy_guard_conditions examples_rclpy_minimal_action_client examples_rclpy_minimal_action_server examples_rclpy_minimal_client examples_rclpy_minimal_publisher examples_rclpy_minimal_service examples_rclpy_minimal_subscriber examples_rclpy_pointcloud_publisher examples_tf2_py foonathan_memory_vendor google_benchmark_vendor gz_cmake_vendor gz_math_vendor gz_utils_vendor iceoryx_posh launch launch_pytest launch_ros launch_testing launch_testing_examples launch_testing_ros launch_xml launch_yaml lifecycle_py mimick_vendor osrf_pycommon qt_dotgraph qt_gui_cpp quality_of_service_demo_py rclpy ros2action ros2bag ros2cli ros2component ros2doctor ros2interface ros2launch ros2lifecycle ros2multicast ros2node ros2param ros2pkg ros2run ros2service ros2test ros2topic ros2trace rosbag2_examples_py rosbag2_performance_benchmarking rosbag2_performance_benchmarking_msgs rosbag2_py rosidl_cli rosidl_pycommon rosidl_runtime_py rpyutils rqt rqt_action rqt_bag rqt_bag_plugins rqt_console rqt_graph rqt_gui rqt_gui_py rqt_msg rqt_plot rqt_publisher rqt_py_console rqt_reconfigure rqt_service_caller rqt_shell rqt_srv rqt_topic rviz_common rviz_ogre_vendor sensor_msgs_py sros2 test_launch_ros test_ros2trace test_tracetools test_tracetools_launch tf2_ros_py tf2_tools topic_monitor tracetools_launch tracetools_read tracetools_test tracetools_trace zenoh_cpp_vendor
root@tomoyafujita:~/ros2_kilted# source install/local_setup.bash
root@tomoyafujita:~/ros2_kilted# ros2 run demo_nodes_cpp talker
[INFO] [1746164369.402888627] [talker]: Publishing: 'Hello World: 1'
[INFO] [1746164370.402681471] [talker]: Publishing: 'Hello World: 2'
[INFO] [1746164371.402674434] [talker]: Publishing: 'Hello World: 3'
[INFO] [1746164372.402642606] [talker]: Publishing: 'Hello World: 4'
^C[INFO] [1746164372.425121702] [rclcpp]: signal_handler(signum=2)
[ros2run]: Received signal:  Interrupt
[INFO] [1746164372.425294672] [rclcpp]: signal_handler(signum=2)
```

closed 2025-05-02 by fujitatomoya