ros2/kilted_tutorial_party #70: Check ROS 2 Pkg CLI

tfoote, 2025-05-01:
## Setup
- Rmw: Fastdds
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check create**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 pkg create my-pkg
       ```
  
    **Expect**
    
    1.
       Creates a folder named `my-pkg`, which has a `package.xml` file, `CMakeLists.txt` file, `src` folder, and `include` folder.
    
  </details>
- [ ] **Check executables**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 pkg executables turtlesim
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 1
       turtlesim draw_square
       turtlesim mimic
       turtlesim turtle_teleop_key
       turtlesim turtlesim_node
       ```
  </details>
- [ ] **Check list**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 pkg list
       ```
  
    **Expect**
    
    1.
       Lists all the packages, for example.
       ```bash
       # stdout in terminal 1
       action_msgs
       action_tutorials_cpp
       action_tutorials_interfaces
       ...
       visualization_msgs
       yaml_cpp_vendor
       zstd_vendor
       ```
  </details>
- [ ] **Check prefix**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 pkg prefix turtlesim
       ```
  
    **Expect**
    
    1.
       This varies depending on your install type. For example, from binaries on Ubuntu, it will be `/opt/ros/iron`
    
  </details>
- [ ] **Check xml**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 pkg xml turtlesim
       ```
  
    **Expect**
    
    1.
       The full `package.xml` of the `turtlesim` package is printed to stdout.
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

ShahazadAbdulla, 2025-05-03:
I'll test this on Windows 11 / amd64 / binary / Fast DDS.

ShahazadAbdulla, 2025-05-03:
Successfully completed all checks for Issue #70 on Windows 11 / amd64 / binary installation using Fast DDS (`rmw_fastrtps_cpp`).

All commands executed successfully and the outputs matched the expectations:
- `ros2 pkg create`: Created package structure as expected.
- `ros2 pkg executables turtlesim`: Listed correct executables.
*   `ros2 pkg list`: Displayed a full list of packages.
*   `ros2 pkg prefix turtlesim`: Returned the correct installation prefix (`C:\pixi_ws\ros2-windows` in my case).
*   `ros2 pkg xml turtlesim`: Printed the full `package.xml` content.

<details>
<summary>Full Terminal Output</summary>
```
(pixi_ros2_rolling) C:\Users\jztsh\ros2_test_temp>ros2 pkg create my-pkg
going to create a new package
package name: my-pkg
destination directory: C:\Users\jztsh\ros2_test_temp
package format: 3
version: 0.0.0
description: TODO: Package description
maintainer: ['ShahazadAbdulla [Shahazad.Abdulla.Engineer@gmail.com](https://www.google.com/url?sa=E&q=mailto%3AShahazad.Abdulla.Engineer%40gmail.com)']
licenses: ['TODO: License declaration']
build type: ament_cmake
dependencies: []
creating folder .\my-pkg
creating .\my-pkg\package.xml
creating source and include folder
creating folder .\my-pkg\src
creating folder .\my-pkg\include\my-pkg
creating .\my-pkg\CMakeLists.txt

[WARNING]: Unknown license 'TODO: License declaration'. This has been set in the package.xml, but no LICENSE file has been created.
It is recommended to use one of the ament license identifiers:
Apache-2.0
BSL-1.0
BSD-2.0
BSD-2-Clause
BSD-3-Clause
GPL-3.0-only
LGPL-3.0-only
MIT
MIT-0

(pixi_ros2_rolling) C:\Users\jztsh\ros2_test_temp>ls my-pkg
CMakeLists.txt include package.xml src

(pixi_ros2_rolling) C:\Users\jztsh\ros2_test_temp>ros2 pkg executables turtlesim
turtlesim draw_square.exe
turtlesim mimic.exe
turtlesim turtle_teleop_key.exe
turtlesim turtlesim_node.exe

(pixi_ros2_rolling) C:\Users\jztsh\ros2_test_temp>ros2 pkg list
action_msgs
action_tutorials_cpp
[... MANY PACKAGES ...]
yaml_cpp_vendor
zstd_vendor

(pixi_ros2_rolling) C:\Users\jztsh\ros2_test_temp>ros2 pkg prefix turtlesim
C:\pixi_ws\ros2-windows

(pixi_ros2_rolling) C:\Users\jztsh\ros2_test_temp>ros2 pkg xml turtlesim
<package format="3">
<name>turtlesim</name>
<version>1.9.2</version>
<description>
turtlesim is a tool made for teaching ROS and ROS packages.
</description>

<maintainer email="alejandro@openrobotics.org">Alejandro Hernandez Cordero</maintainer>

<license>BSD</license>

<url type="website">[http://www.ros.org/wiki/turtlesim](https://www.google.com/url?sa=E&q=http%3A%2F%2Fwww.ros.org%2Fwiki%2Fturtlesim)</url>
<url type="bugtracker">[https://github.com/ros/ros_tutorials/issues](https://www.google.com/url?sa=E&q=https%3A%2F%2Fgithub.com%2Fros%2Fros_tutorials%2Fissues)</url>
<url type="repository">[https://github.com/ros/ros_tutorials](https://www.google.com/url?sa=E&q=https%3A%2F%2Fgithub.com%2Fros%2Fros_tutorials)</url>

<author email="audrow@openrobotics.org">Audrow Nash</author>
<author email="dthomas@osrfoundation.org">Dirk Thomas</author>
<author>Josh Faust</author>
<author email="mabel@openrobotics.org">Mabel Zhang</author>
<author email="michael.jeronimo@openrobotics.org">Michael Jeronimo</author>
<author email="sloretz@openrobotics.org">Shane Loretz</author>

<build_depend>qt5-qmake</build_depend>
<build_depend>qtbase5-dev</build_depend>

<buildtool_depend>ament_cmake</buildtool_depend>

<exec_depend>libqt5-core</exec_depend>
<exec_depend>libqt5-gui</exec_depend>

<depend>ament_index_cpp</depend>
<depend>geometry_msgs</depend>
<depend>rclcpp</depend>
<depend>rclcpp_action</depend>
<depend>std_msgs</depend>
<depend>std_srvs</depend>
<depend>turtlesim_msgs</depend>

<test_depend>ament_lint_auto</test_depend>
<test_depend>ament_lint_common</test_depend>

<export>
<build_type>ament_cmake</build_type>
</export>
</package>
```
</details>
```
*(Note: I've slightly abbreviated the `ros2 pkg list` output in the comment for brevity*
```

ahcorde, 2025-05-05:
Thank you for testing

closed 2025-05-05 by ahcorde