ros2/kilted_tutorial_party #72: Check ROS 2 Pkg CLI

tfoote, 2025-05-01:
## Setup
- Rmw: Zenoh
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Checks
- [ ] **Check create**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 pkg create my-pkg
       ```
  
    **Expect**
    
    1.
       Creates a folder named `my-pkg`, which has a `package.xml` file, `CMakeLists.txt` file, `src` folder, and `include` folder.
    
  </details>
- [ ] **Check executables**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 pkg executables turtlesim
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 1
       turtlesim draw_square
       turtlesim mimic
       turtlesim turtle_teleop_key
       turtlesim turtlesim_node
       ```
  </details>
- [ ] **Check list**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 pkg list
       ```
  
    **Expect**
    
    1.
       Lists all the packages, for example.
       ```bash
       # stdout in terminal 1
       action_msgs
       action_tutorials_cpp
       action_tutorials_interfaces
       ...
       visualization_msgs
       yaml_cpp_vendor
       zstd_vendor
       ```
  </details>
- [ ] **Check prefix**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 pkg prefix turtlesim
       ```
  
    **Expect**
    
    1.
       This varies depending on your install type. For example, from binaries on Ubuntu, it will be `/opt/ros/iron`
    
  </details>
- [ ] **Check xml**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 pkg xml turtlesim
       ```
  
    **Expect**
    
    1.
       The full `package.xml` of the `turtlesim` package is printed to stdout.
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

JatinVira, 2025-05-01:
Ill check this out.
Please assign to me

JatinVira, 2025-05-01:
Happy to take this up

JatinVira, 2025-05-02:
**Check create successful** 
```
root@0514e77f5a3b:/home/ubuntu# tree
.
`-- my-pkg
    |-- CMakeLists.txt
    |-- include
    |   `-- my-pkg
    |-- package.xml
    `-- src

5 directories, 2 files
```

**Check Executable successful**
```
root@0514e77f5a3b:/home/ubuntu# # User input in terminal 1
ros2 pkg executables turtlesim
turtlesim draw_square
turtlesim mimic
turtlesim turtle_teleop_key
turtlesim turtlesim_node
root@0514e77f5a3b:/home/ubuntu# 
```

 **Check List successful**
```
ros2 pkg list
action_msgs
action_tutorials_cpp
action_tutorials_py
actionlib_msgs
ament_cmake
ament_cmake_auto
ament_cmake_copyright
ament_cmake_core
ament_cmake_cppcheck
.....
```

**Check Prefix successful**
```
root@0514e77f5a3b:/home/ubuntu# # User input in terminal 1
ros2 pkg prefix turtlesim
/opt/ros/kilted
```

**Check Xml successful**
```
root@0514e77f5a3b:/home/ubuntu# # User input in terminal 1
ros2 pkg xml turtlesim
<package format="3">
  <name>turtlesim</name>
  <version>1.9.2</version>
  <description>
    turtlesim is a tool made for teaching ROS and ROS packages.
  </description>

  <maintainer email="alejandro@openrobotics.org">Alejandro Hernandez Cordero</maintainer>

  <license>BSD</license>
.....
```

**_All checks successful 
Looks good to me_**

ahcorde, 2025-05-02:
Thank you for testing

closed 2025-05-02 by ahcorde