ros2/kilted_tutorial_party #74: Check ROS 2 run CLI

tfoote, 2025-05-01:
## Setup
- Rmw: Cyclone
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Checks
- [ ] **Check run**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       You'll have to kill this command with Ctrl-C.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_cpp talker
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 1
       [INFO] [1649123362.906256690] [talker]: Publishing: 'Hello World: 1'
       [INFO] [1649123363.906469087] [talker]: Publishing: 'Hello World: 2'
       [INFO] [1649123364.905960807] [talker]: Publishing: 'Hello World: 3'
       ...
       ```
  </details>
- [ ] **Check run with prefix**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run --prefix 'gdb -ex run' demo_nodes_cpp talker
       ```
  
    **Expect**
    
    1.
       This runs the talker node in gdb. Once the talker is running, you can hit Ctrl-C to get to the console and then type `q` to quit gdb.
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

suchetanrs, 2025-05-02:
```
root@cluttered:~# source /opt/ros/kilted/setup.bash                                                                                     [0/0]
root@cluttered:~# export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp         
root@cluttered:~# # User input in terminal 1
ros2 run demo_nodes_cpp talker
[INFO] [1746215686.552720585] [talker]: Publishing: 'Hello World: 1'
[INFO] [1746215687.552702817] [talker]: Publishing: 'Hello World: 2'

```

With GDB:

```
root@cluttered:~# source /opt/ros/kilted/setup.bash
root@cluttered:~# export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
root@cluttered:~# ros2 run --prefix 'gdb -ex run' demo_nodes_cpp talker
GNU gdb (Ubuntu 15.0.50.20240403-0ubuntu1) 15.0.50.20240403-git
Copyright (C) 2024 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.
--Type <RET> for more, q to quit, c to continue without paging--

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /opt/ros/kilted/lib/demo_nodes_cpp/talker...
(No debugging symbols found in /opt/ros/kilted/lib/demo_nodes_cpp/talker)
Starting program: /opt/ros/kilted/lib/demo_nodes_cpp/talker 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
warning: could not find '.gnu_debugaltlink' file for /lib/x86_64-linux-gnu/liblttng-ust.so.1
warning: could not find '.gnu_debugaltlink' file for /lib/x86_64-linux-gnu/liblttng-ust-tracepoint.so.1
[New Thread 0x7ffff740e6c0 (LWP 1343462)]
[New Thread 0x7ffff6c0d6c0 (LWP 1343463)]
[New Thread 0x7ffff5ace6c0 (LWP 1343485)]
[New Thread 0x7ffff52cd6c0 (LWP 1343486)]
[New Thread 0x7ffff4a786c0 (LWP 1343487)]
[New Thread 0x7ffff42776c0 (LWP 1343488)]
[New Thread 0x7ffff3a766c0 (LWP 1343489)]
[New Thread 0x7ffff32756c0 (LWP 1343490)]
[New Thread 0x7ffff29736c0 (LWP 1343491)]
[New Thread 0x7ffff20716c0 (LWP 1343492)]
[New Thread 0x7ffff18706c0 (LWP 1343493)]
[INFO] [1746215798.971344472] [talker]: Publishing: 'Hello World: 1'
[INFO] [1746215799.971322105] [talker]: Publishing: 'Hello World: 2'
[INFO] [1746215800.971178971] [talker]: Publishing: 'Hello World: 3'
^C
Thread 1 "talker" received signal SIGINT, Interrupt.
0x00007ffff790ed71 in __futex_abstimed_wait_common64 (private=0, cancel=true, abstime=0x7fffffffbb20, op=393, expected=0, 
    futex_word=0x5555555ae918) at ./nptl/futex-internal.c:57
warning: 57     ./nptl/futex-internal.c: No such file or directory
(gdb) q
A debugging session is active.

        Inferior 1 [process 1343449] will be killed.

Quit anyway? (y or n) y
root@cluttered:~# 
```

ahcorde, 2025-05-02:
Thank you for testing

closed 2025-05-02 by ahcorde